Fraunhofer-FIT-DIEN/iec104-python #7: TLS Connection Handshake failes

MilanDeruelle, 2024-02-02:
Hello Everyone!

Can anyone get a TLS connection working? 

I'm using the minimal example with added 
```
    tls = c104.TransportSecurity(validate=True, only_known=False)
    tls.set_ca_certificate(cert="./root.cer")
    tls.set_certificate(cert="./client1.cer", key="./client1-key.pem")

    tls.add_allowed_remote_certificate(cert="./server.cer")
```
configuration (for server and client respectively) using the example certificates from the lib60870 tls examples. 
No matter what I configure or how I set it up (even with validate=False) I keep getting 

`TransportSecurity.event] Alarm: Certificate verification failed (t: 2, c: 6, version: TLS 1.2 remote-ip: 127.0.0.1:2404)`
on the client and and on the server
`TransportSecurity.event] Alarm: handshake failed for unknown reason (t: 2, c: 8, version: TLS 1.2 remote-ip: 127.0.0.1:35534)`

I'm not sure if this is a bug in the software or if I'm missing something crucial, but since the documentation of the python part of the library regarding TLS is quite limited I'm running out of ideas on what to try.

Any help/ideas would be appreciated! 

Best Regards!
Milan

MilanDeruelle, 2024-02-02:
Hey, thanks a lot for the quick answer. The generated certs work fine. I'll try if the lib60870 c version as well.

m-unkel, 2024-02-03:
I tested the tests/client.py and tests/server.py successfully against the example certificates provided in lib60870-C repository.

### client.py
```python
    tlsconf = c104.TransportSecurity(validate=True, only_known=True)
    tlsconf.set_certificate(cert=str(ROOT / "certs/client1.cer"), key=str(ROOT / "certs/client1-key.pem"))
    tlsconf.set_ca_certificate(cert=str(ROOT / "certs/root.cer"))
    tlsconf.add_allowed_remote_certificate(cert=str(ROOT / "certs/server.cer"))
```

### server.py
```python
    tlsconf = c104.TransportSecurity(validate=True, only_known=True)
    tlsconf.set_certificate(cert=str(ROOT / "certs/server.cer"), key=str(ROOT / "certs/server-key.pem"))
    tlsconf.set_ca_certificate(cert=str(ROOT / "certs/root.cer"))
    tlsconf.add_allowed_remote_certificate(cert=str(ROOT / "certs/client1.cer"))
```

closed 2024-02-10 by m-unkel

harshcht, 2025-07-21:
I know the issue is closed , but am facing the same error again . generated certificates using the script, but am getting this error on server side : 

`CL] Connection State Changed 123 | State ConnectionState.CLOSED_AWAIT_RECONNECT
CL] Waiting for connection to 127.0.0.1:19998
[c104.Connection] connect] Asynchronous connect to 127.0.0.1:19998
[c104.Connection] state] CLOSED_AWAIT_RECONNECT -> CLOSED_AWAIT_OPEN
[c104.Connection] CALLBACK on_state_change
CL] Connection State Changed 123 | State ConnectionState.CLOSED_AWAIT_OPEN
TLS: handshake failed - mbedtls_ssl_handshake returned -0x5d80
[c104.TransportSecurity] Alarm: handshake failed for unknown reason (t: 2, c: 8, version: TLS 1.3 remote-ip: 127.0.0.1:19998)
[c104.Connection] state] CLOSED_AWAIT_OPEN -> CLOSED_AWAIT_RECONNECT
[c104.Connection] set_closed] Connection closed to 127.0.0.1:19998
[c104.Connection] connection_handler] Connection FAILED to 127.0.0.1:19998 | TOTAL 63 µs
[c104.Connection] CALLBACK on_state_change
CL] Connection State Changed 123 | State ConnectionState.CLOSED_AWAIT_RECONNECT
^CTraceback (most recent call last):`
and this on the client side : 
`TLS: handshake failed - mbedtls_ssl_handshake returned -0x1
[c104.TransportSecurity] Alarm: handshake failed for unknown reason (t: 2, c: 8, version: TLS 1.3 remote-ip: (null))
[c104.Server] CALLBACK on_connect
SV] <->| 127.0.0.1 | SERVER 0.0.0.0:19998
TLS: handshake failed - mbedtls_ssl_handshake returned -0x1`

I am testing the clinet..py and server.py scripts in tests folder directly .