Fraunhofer-FIT-DIEN/iec104-python #10: Unable to install c104 package

nekittheragdoll, 2024-02-06:
Hello!

I'm attempting to freshly install the package of the current version 1.17.1. However, during the installation process an error appears:
<details>
<summary>Error message</summary>

```
Collecting c104
  Using cached c104-1.17.1.tar.gz (21 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: c104
  Building wheel for c104 (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for c104 (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [70 lines of output]
      running bdist_wheel
      running build
      running build_ext
      CMake Warning:
        Ignoring extra path from command line:

         "<omitted>\AppData\Local\Temp\pip-install-fsnogc9u\c104_ac3d6c9f0b6448bcb87ad28f9437a588"

      CMake Error: The source directory "<omitted>/AppData/Local/Temp/pip-install-fsnogc9u/c104_ac3d6c9f0b6448bcb87ad28f9437a588" does not appear to contain CMakeLists.txt.
      Specify --help for usage, or press the help button on the CMake GUI.
      Traceback (most recent call last):
        File "<omitted>\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "<omitted>\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "<omitted>\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 251, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\build_meta.py", line 404, in build_wheel
          return self._build_with_temp_dir(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\build_meta.py", line 389, in _build_with_temp_dir
          self.run_setup()
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 131, in <module>
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\core.py", line 185, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\core.py", line 201, in run_commands
          dist.run_commands()
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\dist.py", line 963, in run_command
          super().run_command(command)
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\wheel\bdist_wheel.py", line 368, in run
          self.run_command("build")
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\dist.py", line 963, in run_command
          super().run_command(command)
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\command\build.py", line 131, in run
          self.run_command(cmd_name)
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\dist.py", line 963, in run_command
          super().run_command(command)
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\command\build_ext.py", line 88, in run
          _build_ext.run(self)
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\command\build_ext.py", line 345, in run
          self.build_extensions()
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\command\build_ext.py", line 467, in build_extensions
          self._build_extensions_serial()
        File "<omitted>\AppData\Local\Temp\pip-build-env-r9yoj7_l\overlay\Lib\site-packages\setuptools\_distutils\command\build_ext.py", line 493, in _build_extensions_serial
          self.build_extension(ext)
        File "<string>", line 123, in build_extension
        File "<omitted>\AppData\Local\Programs\Python\Python312\Lib\subprocess.py", line 571, in run
          raise CalledProcessError(retcode, process.args,
      subprocess.CalledProcessError: Command '['cmake', '<omitted>\\AppData\\Local\\Temp\\pip-install-fsnogc9u\\c104_ac3d6c9f0b6448bcb87ad28f9437a588', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=<omitted>\\AppData\\Local\\Temp\\pip-install-fsnogc9u\\c104_ac3d6c9f0b6448bcb87ad28f9437a588\\build\\lib.win-amd64-cpython-312\\', '-DPYTHON_EXECUTABLE=<omitted>\\AppData\\Local\\Programs\\Python\\Python312\\python.exe', '-DCMAKE_BUILD_TYPE=Release', '-DC104_VERSION_INFO=1.17.1', '-A', 'x64', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY_RELEASE=<omitted>\\AppData\\Local\\Temp\\pip-install-fsnogc9u\\c104_ac3d6c9f0b6448bcb87ad28f9437a588\\build\\lib.win-amd64-cpython-312']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for c104
Failed to build c104
ERROR: Could not build wheels for c104, which is required to install pyproject.toml-based projects
```
</details>

I tried to install the package on Windows, Ubuntu, both Python312 and Python 311. All attempts produce same error.

Note:
Few months ago I tested the package version 1.16.1 and it was installed and worked without problems, therefore I'm trying to figure out if the issue's on my side or in the package distribution.

m-unkel, 2024-02-07:
Hey,
thanks for the report.

Concerning Python 311 there are pre-build binary wheels for Windows and Ubuntu available via pypi.org / pip.
Python 312 support was not yet tested, I will have a look into that.

Maybe it is a problem related to the source distribution via pypi, I will check that and report back.
Could you please test wether this works:
pip install git+https://github.com/Fraunhofer-FIT-DIEN/iec104-python.git@v1.17.1
(requires the git executable to be available on your system)

m-unkel, 2024-02-08:
Should be fixed.
I uploaded a new source distribution to pypi.

nekittheragdoll, 2024-02-11:
Thank you, it works now. Not sure if intended, but I was able to install it and run it on Python312. I tested only pypi though.

closed 2024-02-13 by m-unkel