Fraunhofer-FIT-DIEN/iec104-python #9: timeout on transmit

jeremykalvar, 2024-02-06:
Hello, 
Thank you for this repo, very useful!

I have implemented a Client that needs to transmit periodically some points to a remote server. 
Is there a way to ignore the server response on tranmsits ? I keep on getting failed responses that seem to block the transmits for a few seconds, and yet the server receives the points correctly.

```python
import functools
import threading
import time

import c104

from iec.handler import cl_on_new_point, cl_on_new_station
from instance.config import get_settings

def cl_pt_on_receive_point(
    point: c104.Point, previous_state: dict, message: c104.IncomingMessage
) -> c104.ResponseState:
    return c104.ResponseState.SUCCESS

def cl_on_new_station(
    client: c104.Client, connection: c104.Connection, common_address: int
) -> None:
    connection.add_station(common_address=common_address)

def cl_on_new_point(
    client: c104.Client, station: c104.Station, io_address: int, point_type: c104.Type
) -> None:
    point = station.add_point(io_address=io_address, type=point_type)
    point.on_receive(callable=cl_pt_on_receive_point)

class IECClient:
    def __init__(
        self,
        ip,
        port,
        originator_address=132,
        tick_rate_ms=1000,
        command_timeout_ms=5000,
        common_address=27808,
    ):
        self.ip = ip
        self.port = port
        self.common_address = common_address
        self.client = c104.Client(tick_rate_ms=tick_rate_ms, command_timeout_ms=command_timeout_ms)
        self.client.originator_address = originator_address
        self.connection = self.client.add_connection(
            ip=self.ip, port=self.port, init=c104.Init.INTERROGATION
        )

        self.client.on_new_station(callable=functools.partial(cl_on_new_station))
        self.client.on_new_point(callable=cl_on_new_point)
        self.station = self.connection.add_station(common_address=self.common_address)

    def stop(self):
        self.client.stop()

    def start(self):
        self.client.start()
        while not self.connection.is_connected:
            print(f"Waiting for connection to {self.connection.ip}:{self.connection.port}")
            time.sleep(3)

    def send_n_level(self, ioa: int, value: float):
        point = self.station.get_point(io_address=ioa)
        if not point:
            point = self.station.add_point(io_address=ioa, type=c104.Type.C_SE_NC_1)
        point.value = value

        transmit_thread = threading.Thread(target=self._transmit_point, args=(point, ioa, value))
        transmit_thread.start()

    def _transmit_point(self, point, ioa, value):
        if point.transmit(cause=c104.Cot.ACTIVATION):
            print("SUCCESS")
        else:
            print("FAILURE")

    def send_cg(self):
        interrogation = self.connection.interrogation(common_address=self.common_address)
        print(f"Interrogation: {interrogation}")

iec_master_client = IECClient(ip=settings.IEC_SERVER_HOST, port=settings.IEC_SERVER_PORT)
```

m-unkel, 2024-02-07:
Hey,
first of all the transmission should be fast enough to wait for the response, but technically it would be possible to add an argument to ignore the response.
How many commands do you want to send per second?
Maybe there is an undesired locking problem.
Does the problem occur only by sending the same point multiple times or also by sending different points at the same time?

Your code does not show how send_point(...) is called.
But your thread calls transmit in a loop without any sleep.. you are bombing your RTU with command messages, which is not an to be expected behavior unless you want to attack a system. Is that the case?

By the way.. the callbacks you define (cl_pt_on_receive_point, cl_on_new_station cl_on_new_point) reproduce exactly the default behavior in case the callback is not set, so you could remove them unless you want to add additional logic.

jeremykalvar, 2024-02-08:
Hi,
Thanks for the reply!
No my goal is not to attack a system haha... Yet !
Ideally, I would like to be able to send 1 command every second.
Right now I am working on a single point, so yes, the problem seems to occur only by sending the same point multiple times.

I am using it in a fastApi router:

```python
@router.post("/send-iec-point")
    async def send_iec_point(, ioa: int, filename: str):
        value = self.csv_handler.read_next_line_from_csv(filename, ioa)
        iec_master_client.send_point(ioa=ioa, value=float(value))
        return value
``` 

If the endpoint is called more than once every 3 seconds, my app breaks.
I may be missing something...

divinity666, 2024-02-08:
I was also playing with this beautiful library and went to a similar point:

I am also receiving `False` as a response to `transmit`, though I implemented the `on_receive` callback and always return `c104.ResponseState.SUCCESS`. Furthermore, it takes ~3 seconds for the library to proceed after this `transmit` call.

Sounds pretty much, like the same issue.

divinity666, 2024-02-09:
Ok, I found the root cause for me:

I was calling `transmit(cause=c104.Cot.SPONTANEOUS)`. If I change the call to `transmit(cause=c104.Cot.ACTIVATION)` it works as expected.

Nevermind.

m-unkel, 2024-02-09:
> ```python
> class IECClient:
> ```

This is the default behavior, you do not have to implement the callback to return "Success".

> Ok, I found the root cause for me:
> 
> I was calling `transmit(cause=c104.Cot.SPONTANEOUS)`. If I change the call to `transmit(cause=c104.Cot.ACTIVATION)` it works as expected.
> 
> Nevermind.

Yes spontaneous should be used in monitoring direction only AFAIK.

m-unkel, 2024-02-09:
@jeremykalvar could you please generate some logging information? A command+confirmation should consume milliseconds not seconds..

```python

c104.set_debug_mode(mode=c104.Debug.Client|c104.Debug.Connection)

def con_receive_raw(connection: c104.Connection, data: bytes) -> None:
    print("RECV] {1} [{0}] | CONN OA {2}".format(data.hex(), c104.explain_bytes_dict(apdu=data), connection.originator_address))

def con_send_raw(connection: c104.Connection, data: bytes) -> None:
    print("SEND] <--| {1} [{0}] | CONN OA {2}".format(data.hex(), c104.explain_bytes_dict(apdu=data), connection.originator_address))

class IECClient:
    def __init__(...):
        # [...] 
        # append the following to your __init__
        self.connection.on_receive_raw(callable=con_receive_raw)
        self.connection.on_send_raw(callable=con_send_raw)

```

closed 2024-03-20 by m-unkel