Fraunhofer-FIT-DIEN/iec104-python #11: Wheel problem

Temuulen-the-maker, 2024-02-29:
I wanted python based c104 emulator. This package looked exactly what I needed. But I run into installation problem. The exact error message is as follows:

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for c104
Failed to build c104
ERROR: Could not build wheels for c104, which is required to install pyproject.toml-based projects

My python and pip should be latest updated versions of their kind.
Attached snip of the error. How to solve 'wheel' problem? I know they say do not reinvent wheel, use it. But I cannot even have this wheel.

[image]

m-unkel, 2024-03-03:
You can try installing the latest binary distribution `pip install c104==1.17.1`.
The 1.17.1.postX Version does not contain the windows binary - i will fix this soon.

The error seems to be related to the version of the MSVC Build Tools. I will look into the topic next week, but hopefully the solution above solves your problem.

Temuulen-the-maker, 2024-03-08:
Hello Mr.Unkel,

The suggested line of command works when there is no environment activated in the terminal. 
But Visual studio needs a kernel or an virtual environment. The problem is the line of code (....1.17.1) does not work when I activated an environment. It throws this error (errorMessage img):
[image]

And strangely, the line "python3 -m pip install c104" returns positive. But still doesnt work:
[image]

I would really appreciate the fix for windows.
Out of desparity, I am thinking to install second OS of LINUX on my machine. I see the installation is smooth on Linux. Is it advisable?

m-unkel, 2024-03-10:
Looks like your system installation is python 3.11 and the virtual environment uses python 3.12 that will only work with `c104==1.17.1.post6`
It is not necessary to switch the OS.
I made a mistake by uploading these 1.17.1.postX patches that do not contain all kind of pre-build binaries.

This problem should be fixed with next release 1.8.0 that will be available within the next 24 hours, because it will contain the pre-build binaries for all platforms and architectures in the latest release and you should be able to simply use `pip install c104` in your system python installation as well as the virtual environment again. 
Sorry for the confusion.

m-unkel, 2024-03-11:
Version 1.8.0 has been released. Please provide feedback on whether the issue has been resolved.

closed 2024-03-20 by m-unkel

Temuulen-the-maker, 2024-03-25:
Hello Mr Unkel,

The c104 installation worked for me. Thanks for the help.

But I want to ask how they solve it in practice to have a client-client
connection or server-server connection or a station that is both server and
client depending on situations? I could use any literature/ source/
explanation for this matter.

Actually in c104 documentation
<https://iec104-python.readthedocs.io/python/index.html>, I only see
client-server connections. This means, from a programming point of view, it
is not supported to have above types of connections?

Thanks for continuous support!

m-unkel, 2024-04-08:
This type of connection is not intended by the protocol standard. In general, it is possible to run a server and a client instance on both sides. But what would be the use case for electrical power systems?

Temuulen-the-maker, 2024-04-11:
Hello,

You are right. I have been practicing this c104 and have asked a stupid
question.

But there is 1 more thing I cannot do with the library.

So my client is sending measured values with time tags (M_ME_TB). Problem
is I cannot print the received value with its tagged timestamp in the
python window.

I only find properties, such as, received_at_ms, sent_at_ms and so on.
These timestamps are, as I understand, to mark the time when an action is
taken, such as receiving and sending, respectively.

My need is to read a timestamp that is tagged to the message that was sent
from server to the client. How can I do this? Is my question clear? So this
type of timestamp is, I think, part of the message.

m-unkel, 2024-04-18:
At client side you can use point.updated_at_ms to get the timestamp sent by the server.
If you would also like to set this timestamp manually on the server side, have a look at [point.set()](https://iec104-python.readthedocs.io/python/point.html#c104.Point.set) methods argument timestamp_ms.

Temuulen-the-maker, 2024-07-22:
Hello Mr. Unkel,

I came to you to ask a question. I have a 60870 server running. The server runs on simulation platform (Typhoon *link below) sends out some messages with timestamps. But I have developed a client to communicate with that server.

My problem is I cannot print those timestamp on python client side. Can you help me here? thanks

Link:
https://www.typhoon-hil.com/

Temuulen-the-maker, 2024-07-22:
# comm_obj.py
import c104
import time

client = None
connection = None
station = None
server = None
inp = None

# ------------------------------------------------------------------------------------------------------------------------------------------------------------

def cl_obj(tick_rate_ms, command_timeout_ms):
    client = c104.Client(tick_rate_ms = tick_rate_ms, command_timeout_ms = command_timeout_ms)
    return client

def cl_con(client, ip, port, init):
    connection = client.add_connection(ip = ip, port = port, init = init)
    return connection

def cl_sta(connection, common_address):
    station = connection.add_station(common_address = common_address)
    return station

def cl_take(station, io_address, type):
    point = station.add_point(io_address = io_address, type = type)
    return point

def cl_send(station, io_address, type):
    point = station.add_point(io_address = io_address, type = type)
    return point

def con_on_send_raw(connection: c104.Connection, data: bytes) -> None:
    print("<--| {1} [{0}] | CON {2}:{3}".format(data.hex(), c104.explain_bytes(apdu=data), connection.ip, connection.port))

def con_check(connection, common_address, cause, qualifier):
    return connection.interrogation(common_address = common_address, cause = cause, qualifier = qualifier)

# ------------------------------------------------------------------------------------------------------------------------------------------------------------

def sv_obj(ip, port, max_connections): # server object
    server = c104.Server(ip = ip, port = port, max_connections = max_connections)
    return server

def sv_sta(server, CA):
    station = server.add_station(common_address = CA)
    return station

def sv_take(station, IOA, type): # server command message = a message server receives as a command
    point = station.add_point(io_address = IOA, type = type)
    return point

def sv_send(station, IOA, type, report_ms): # server monitor message = a message server sends as a report
    point = station.add_point(io_address = IOA, type = type, report_ms = report_ms)
    return point

def sv_on_connect(server: c104.Server, ip: str) -> bool:
        # print("connected")
        pass
def ser_check(server):
    def sv_on_connect(server: c104.Server, ip: str) -> bool:
            # print("connected")
            pass
    return server.on_connect(callable = sv_on_connect)
# ------------------------------------------------------------------------------------------------------------------------------------------------------------

def sl(inp):
    time.sleep(inp)

def up1(inp):
    return inp + 1

Temuulen-the-maker, 2024-07-22:
# clientFile.py

import comm_obj as co
import c104
import time
import random

'''
Proof of intended operation:
[ ] check if 1 typhoon server can connect to 6 python clients
[ ] check if internal communication with agent works as expected
    prerequisite: must get agent from Jens.
[ ] check time syncronization via ptp: 'ptp enable'
[ ]

Other things to check:
[ ] check if init = CLOCK_SYNC synchronizes the clock between client and server, if yes does it report/print some confirmation/information
[ ] investigate time stamp option from typhoon 104 server block
[ ] check if unmatching CoT, qualifier between typhoon server and python client can work
[+] know what is communication stopping procedure
    never stops, but it is just persistant client, it always listens to the network
[ ] know what is communication starting procedure
[ ] TIMESTAMP: check this command: sv_measurement_point.set(value=-12.56, quality=c104.Quality.Invalid, timestamp_ms=int(time.time() * 1000))
[ ] SERVER:on_clock_sync: check if this synchs clock

Problems:
[ ] this client does not recognize when server is down - BUT CLIENT 2 RECOGNIZES ALTHOUGHT CLIENT 1 AND 2 ARE IDENTICAL
[ ] when two clients connected, there is intermediary disconnection: there is a disruption in ~7 loops

Observations:
- with Typhoon, single connection can work with client that has if connection.interrogation(), but multiple client connections can work if connection.is_connected
- client with bigger common address recognizes when server is down
- when a client runs alone, the client does not recognize when server is down
- 

'''

def con_on_receive_raw(connection: c104.Connection, data: bytes) -> None:
    print("-->| {1} [{0}] | CON {2}:{3}".format(data.hex(), c104.explain_bytes(apdu=data), connection.ip, connection.port))

# general parameterization:
slptm = 1 # sleeptime in seconds
num = 500

# client parameterization:
tick_rate_ms        = num                      # client thread update interval
command_timeout_ms  = num                        # time to wait for a command to response
ip                  = "192.168.42.120"           # typhoon
# ip                  = "10.49.23.147"             # OFFIS
port                = 2404                      # remote terminal units port
init                = c104.Init.INTERROGATION   # communication initiation command # ALL:0, CLOCL_SYNC:2, INTERROGATION:1, NONE:3; properties init.name, init.value
common_address      = 2
io_address          = 1
cl_takeType         = c104.Type.M_ME_NB_1 # = sv_sendType from Typhoon  # Measured value, scaled value
cl_sendType         = c104.Type.C_SE_NB_1 # = sv_takeType to Typhoon    # Setpoint command, scaled value
cause               = c104.Cot.ACTIVATION
qualifier           = c104.Qoi.GROUP_1

# client object creation
client              = co.cl_obj(tick_rate_ms, command_timeout_ms)
connection          = co.cl_con(client, ip, port, init)
station             = co.cl_sta(connection, common_address)

cl_take             = co.cl_take(station, io_address, cl_takeType)
cl_send             = co.cl_send(station, io_address + 1, cl_sendType)

# running the client
counter = 1
client.start()
try:
    while client.is_running:
        if connection.is_connected:
            # from server
            connection.interrogation(common_address, cause, qualifier)
            print(counter, ": received message :", cl_take.value)

            # to server
            cl_send.value = round(random.random(),2) * 100
            cl_send.transmit()
            print(counter, ": sent message :", cl_send.value)

        else:
            print('client is not connected --------------------------------------------------')
            pass    

        print(counter, ': =====================')
        counter += 1
        time.sleep(slptm)
except KeyboardInterrupt:
    print("--- client stopped successfully ---")
    client.stop()
    pass
client.stop()

Temuulen-the-maker, 2024-07-22:
I just pasted my python files here, in case you could spot a problem in my code.

I do not have much experience with github, the pasted code looks strange.

m-unkel, 2024-07-23:
Hey, i wrapped your code into code blocks to allow syntax highlighting.
You can do it like this:

> \`\`\`python
> 
> code...
>
> \`\`\`

The first things is recognize not regarding the question directly:
- you are using init **c104.Init.INTERROGATION**, but are calling interrogation manually within the loop.\
  This will cause to consecutive interrogations at start. \
  A init **c104.Init.NONE** would be fine in that case as well.
- if you want to read just one point, you could call **cl_take.read()** directly instead of **connection.interrogation(...)**
- use **cl_take.updated_at_ms** to print the timestamp

```python
# clientFile.py

import comm_obj as co
import c104
import time
import random

def con_on_receive_raw(connection: c104.Connection, data: bytes) -> None:
    print("-->| {1} [{0}] | CON {2}:{3}".format(data.hex(), c104.explain_bytes(apdu=data), connection.ip, connection.port))

# general parameterization:
slptm = 1 # sleeptime in seconds
num = 500

# client parameterization:
tick_rate_ms        = num                      # client thread update interval
command_timeout_ms  = num                        # time to wait for a command to response
ip                  = "192.168.42.120"           # typhoon
# ip                  = "10.49.23.147"             # OFFIS
port                = 2404                      # remote terminal units port
init                = c104.Init.NONE   # communication initiation command # ALL:0, CLOCL_SYNC:2, INTERROGATION:1, NONE:3; properties init.name, init.value
common_address      = 2
io_address          = 1
cl_takeType         = c104.Type.M_ME_NB_1 # = sv_sendType from Typhoon  # Measured value, scaled value
cl_sendType         = c104.Type.C_SE_NB_1 # = sv_takeType to Typhoon    # Setpoint command, scaled value
cause               = c104.Cot.ACTIVATION
qualifier           = c104.Qoi.GROUP_1

# client object creation
client              = co.cl_obj(tick_rate_ms, command_timeout_ms)
connection          = co.cl_con(client, ip, port, init)
station             = co.cl_sta(connection, common_address)

cl_take             = co.cl_take(station, io_address, cl_takeType)
cl_send             = co.cl_send(station, io_address + 1, cl_sendType)

# running the client
counter = 1
client.start()
try:
    while client.is_running:
        if connection.is_connected:
            # from server
            if cl_take.read():
                print(f"{counter}: received message: {cl_take.value} measured at {cl_take.updated_at_ms}")

            # to server
            cl_send.value = round(random.random(),2) * 100
            cl_send.transmit()
            print(counter, ": sent message :", cl_send.value)

        else:
            print('client is not connected --------------------------------------------------')
            pass    

        print(counter, ': =====================')
        counter += 1
        time.sleep(slptm)
except KeyboardInterrupt:
    print("--- user requested stop ---")
except Exception as e:
    import traceback
    print(str(e) + "\n" + str(traceback.format_exc()))
finally:
    client.stop()
    print("--- client stopped successfully ---")

```

Temuulen-the-maker, 2024-07-26:
Hello thanks for the improvement to my code.

I have 3 questions:
1. this update_at_ms is I think python side measuring time. It is the same
result as time.time(). What I wanted was a bit different. So from the
server side message is sent, I believe, with a timestamp that marks sending
time. That message travels some network to get to the client, then the
client registers it at a timestamp. The function 'updated_at_ms' is, I
think, that registered time. Is it possible to read a timestamp that was
attached to the message?
2. I tested cl_take.read(). This read method returns False and does not
print the received message. It seems in the while loop I must have
interrogation. I took your advice and am running it on NONE init.
3. General question: I am interested in communication. So I want to ask
what online source I can use to learn more about it? Until you explained
INTERROGATION init is causing consecutive interrogations, I did not know it
was an issue because the code was working and printing the delivered
messages. I want to make sure my code has a bit of practical sense and
correctness.

Thanks for continuous help and support! I try to be less of a distraction
but sometimes I do not have anybody to ask questions. So I come to you.

Temuulen-the-maker, 2024-07-26:
I think it is helpful to explain that my server is located at an energy
network simulator, named Typhoon (link <https://www.typhoon-hil.com/>). I
am not bouncing messages between python client and python server.

m-unkel, 2024-08-12:
Ah now i see the problem, you are using the wrong point type.
There is always one version with and one without a timestamp.
M_ME_NB_1 -> no time info included
M_ME_TE_1 -> with time info

The library injects the received timestamp if it is a type that does not transport a time info itself, therefor you read the python time. And it should not make a difference if you are talking to a SPS or a physical SCADA unit or another python client. Everything support should be compliant.

m-unkel, 2024-08-12:
As a short rule, if the 5th letter of a type is a T it means with timestamp, if it is an N it is a message without a timestamp.
M_xx_Nx_1
M_xx_Tx_1

The same is valid for commands:
C_xx_Nx_1
C_xx_Tx_1