Fraunhofer-FIT-DIEN/iec104-python #13: Sending timestamps not in UTC

ArtTron22, 2024-03-05:
Hello! 

I am trying to adjust the time of a timestamp when sending point data, and I would like to know how to do so. When sending data about a point with the type "type=c 104.Type.M_ME_TF_1", I want the timestamp to be in the local time zone of the server, rather than the UTC time. For example, the server's local time zone is Europe/Istanbul. Could you please help me with this? 

Thank you for your attention.

m-unkel, 2024-03-07:
Hi there, 
since the protocol isn't timezone aware it would be better to use UTC time and instead format the time into a desired timezone in the presentation layer.

But it should be possible to manually set a timestamp for a measurement value via the set method.
point.set(value=123, timestamp_ms=...)
I tried the last option and found a bug there that will be fixed in the upcoming release 1.8.0 within the next days .

ArtTron22, 2024-03-09:
Thank you for your answer. And when do you plan to release version 1.8.0, approximately?

m-unkel, 2024-03-10:
Within the next 24 hours.

m-unkel, 2024-03-11:
Version 1.8.0 has been released. Please provide feedback on whether the issue has been resolved.

ArtTron22, 2024-03-11:
Thank you so much! Great update!

closed 2024-03-11 by ArtTron22