Fraunhofer-FIT-DIEN/iec104-python #15: server on_receive callback: message.raw cannot be accessed

thlor, 2024-04-16:
In my `c104.Server`, I added a `c104.Type.C_SC_NA_1` point, and attached an `on_receive` callback function to it.

In the callback function, I cannot read the `message.raw` property (e.g. simply: `raw = message.raw` or `print(f"value: {message.raw})`). I get the following error:

```
Point.on_receive] Error:
TypeError: Unregistered type : _object

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/server.py", line 100, in _c104_point_shutdown_callback
    raw = message.raw
TypeError: Unable to convert function return value to a Python type! The signature was
  (arg0: c104.IncomingMessage) -> _object
```

The `message.raw_explain` and `message.value` properties can be accessed as expected. The client that transmits the message is also a iec104-python `c104.Client`, running in a separate docker container.

c104 version: 1.18.0
python version: 3.12.2

m-unkel, 2024-04-18:
I can confirm the issue.

To address it promptly, I've published an initial fix. You can test this fix by installing the package directly from the source using the following command:
```shell
pip install git+https://github.com/Fraunhofer-FIT-DIEN/iec104-python.git@15-server-on_receive-callback-messageraw-cannot-be-accessed
```
I must acknowledge that, due to time constraints, I won't be able to release a new version before June.
Thank you for your patience and for your contribution to improving the project!

thlor, 2024-04-19:
@m-unkel thanks very much for the quick response. Great project! :)

m-unkel, 2024-09-03:
Release 2.0.0 is now available, and it includes the fix that addresses this issue. I would greatly appreciate your feedback. If the issue is resolved to your satisfaction, please consider closing this thread.

Thank you!

m-unkel, 2024-09-23:
We have released a fix addressing this issue. Since we haven't heard back from you in a while, we assume the problem has been resolved. As a result, we are closing this issue.

If you're still experiencing any problems or have further questions, please feel free to reopen the issue or create a new one.
Thank you!

closed 2024-09-23 by m-unkel