Fraunhofer-FIT-DIEN/iec104-python #19: Setting the quality value at the default Point

ArtTron22, 2024-05-06:
Hello. I had the following question: imagine that at some point my daughter's quality parameter changed to point.quality = c 104.Quality.Invalid. But then, after several polls, the quality of this point should become normal again, those QUALITY = Quality set: {}, is_good: True. How could I do that?
I tried to change this somehow in the before_transmit function in point.on_before_read(callable=before_transmit), where point.set(value=32, quality=point.quality, timestamp_ms=int(time.time() * 1000)), but I failed. How can I set the default value to quality again?

m-unkel, 2024-05-07:
Hello, you should be able to use the empty constructor to reset the quality to Good.

```python
point.quality = c104.Quality()
```

ArtTron22, 2024-05-08:
Thank you very much! 
point.quality = c104.Quality() works great)

closed 2024-05-08 by ArtTron22