Fraunhofer-FIT-DIEN/iec104-python #22: Fatal error: terminate called after throwing an instance of 'std::bad_weak_ptr'

aklira, 2024-06-13:
I am experiencing an issue with the examples/simple_server.py:
After a short time (few seconds) the iec104 server would crash with the following error:
terminate called after throwing an instance of 'std::bad_weak_ptr'

Here is the full log of the server instance:

```
[c104.Server] Created
[c104.Server] add_station] CA 47
[c104.Server] start] Started
[c104.Server] thread_run] Cannot keep up the tick rate: 73616 �s
[c104.Server] connection_event_handler] Connection OPENED by 192.168.230.196:54704 | TOTAL 34 �s
[c104.Server] raw_message_handler] TOTAL 22 �s
[c104.Server] connection_event_handler] Connection ACTIVATED by 192.168.230.196:54704 | TOTAL 23 �s
[c104.Server] raw_message_handler] TOTAL 0 �s
[c104.Server] raw_message_handler] TOTAL 5 �s
[c104.Server] send_activation_confirmation] to requesting MTU
[c104.Server] raw_message_handler] TOTAL 0 �s
[c104.Server] raw_message_handler] TOTAL 0 �s
[c104.Server] send_interrogation_response] Request | TOTAL 227 �s
[c104.Server] raw_message_handler] TOTAL 0 �s
[c104.Server] interrogation_handler] | IP 192.168.230.196:54704 | OA 0 | CA 47 | TOTAL 344 �s
[c104.Server] thread_run] Connected clients: 1
[c104.Server] send_interrogation_response] Periodic | TOTAL 293 �s
[c104.Server] raw_message_handler] TOTAL 13 �s
[c104.Server] raw_message_handler] TOTAL 8 �s
[c104.Server] raw_message_handler] TOTAL 8 �s
[c104.Server] raw_message_handler] TOTAL 1 �s
[c104.Server] send_interrogation_response] Periodic | TOTAL 384 �s
[c104.Server] raw_message_handler] TOTAL 6 �s
[c104.Server] raw_message_handler] TOTAL 4 �s
[c104.Server] raw_message_handler] TOTAL 6 �s
[c104.Server] raw_message_handler] TOTAL 0 �s
[c104.Server] send_interrogation_response] Periodic | TOTAL 134 �s
[c104.Server] raw_message_handler] TOTAL 7 �s
terminate called after throwing an instance of 'std::bad_weak_ptr'
  what():  bad_weak_ptr
```

m-unkel, 2024-07-23:
I'm sorry, but I cannot reproduce the problem.

Could you please provide more details?
Specifically, how did you install or build c104?
Additionally, how do you execute the script?

aklira, 2024-07-25:
Hi,
I've made the installation following the instructions provided on the Readme:
`python3 -m pip install c104`

I've executed the script on Ubuntu 20.04 VM, x86_64 architecture with python 3.10.

m-unkel, 2024-07-25:
Okay I am able to reproduce the issue.
The issue may happen in server shutdown procedure.
Your client seems to be disconnected and the server.py script quits after disconnect.
Therefore the python garbage collector will free the objects in an not optimal order.
The simple fix is to add a server.stop() command after the last while loop.

```
--- a/examples/simple_server.py
+++ b/examples/simple_server.py
@@ -55,6 +55,8 @@ def main():
         print("Keep alive until disconnected")
         time.sleep(1)
 
+    server.stop()
+
 
 if __name__ == "__main__":
     c104.set_debug_mode(c104.Debug.Server)
```

Impact:
The automatic shutdown procedure is not as clean as it should be, but it should be safe to use the library with calling the stop command. The error should never occur without the python garbage collector is about to free the server instance.

Can you confirm that?

m-unkel, 2024-07-25:
If you wish to, you can verify the fix via installing from source branch:

```shell
pip install git+https://github.com/Fraunhofer-FIT-DIEN/iec104-python.git@dev
```

The fix will be included in the upcoming release.

m-unkel, 2024-09-03:
Release 2.0.0 is now available, and it includes the fix that addresses this issue. I would greatly appreciate your feedback. If the issue is resolved to your satisfaction, please consider closing this thread.

Thank you!

m-unkel, 2024-09-23:
We have released a fix addressing this issue. Since we haven't heard back from you in a while, we assume the problem has been resolved. As a result, we are closing this issue.

If you're still experiencing any problems or have further questions, please feel free to reopen the issue or create a new one.
Thank you!

commit 3c33865d (2024-07-25): :bug: fix #22, improve logging code style
commit 7576edd2 (2024-07-25): :bug: apply shared_from_this fix #22 for client connections
closed 2024-09-23 by m-unkel