Fraunhofer-FIT-DIEN/iec104-python #25: Invalid information object address 0

Geri78-pr, 2024-07-31:
At station.add_point this is writen on the documentation:
io_address ([int](https://docs.python.org/3/library/functions.html#int)) – point information object address (value between 0 and 16777216)

I tried to use io_address=0 and I get error message:
Invalid information object address 0

m-unkel, 2024-07-31:
Hello Geri78-pr,
Thank you for bringing this issue to our attention.

The wording “(value between 0 and 16777216)” was intended to indicate that 0 and 16777216 themselves are not included, only the values in between, which explains the error you encountered when attempting to use io_address=0.

To avoid further confusion, we will update the documentation to clearly indicate that the acceptable range is 1 to 16777215.

Please let us know if you have any other questions or need further clarification.

Geri78-pr, 2024-07-31:
Thank you.

Unfortunately I shoud use 0 io_address also.

Am I correct if I can not do it with your library?

m-unkel, 2024-07-31:
I've pushed a fix to the **dev** branch that now allows the use of **io_address** 0. Previously, it was blocked because we believed the standard only permitted addresses from 1 and up. However, there doesn't appear to be any reason not to allow 0 if other communication partners accept it.

You can verify the fix by installing it from the source branch:

If you wish to, you can verify the fix via installing from source branch:

```shell
pip install git+https://github.com/Fraunhofer-FIT-DIEN/iec104-python.git@dev
```

This change will be included in the upcoming release by the end of the week.

m-unkel, 2024-07-31:
Well i guess it is not a simple as that.. there seems to be problems with the base library https://github.com/mz-automation/lib60870 we are using. I will run more tests and come back to you.

Geri78-pr, 2024-07-31:
Thank you for your effort.
I look forward to this function.

m-unkel, 2024-07-31:
Okay so the latest push should work, but the related_io_address = 0 feature is not yet supported, because it uses 0 as unset value. I will fix this within the next days.

Geri78-pr, 2024-07-31:
Thank you very much.
I try it and give you a feedback soon.

Geri78-pr, 2024-07-31:
I tried the latest dev version with io_address=0 and it seems to be work.
Thank you for helping us so quickly.

I detected something with this version (pevious version did not do it):
With interrogation I got this message (in 2 rows):
[c104.IncomingMessage.extract] Unsupported type C_IC_NA_1
[c104.IncomingMessage.extract] Unsupported type C_IC_NA_1

I could see that the interrogation works and get the data but this message is a bit strange.

m-unkel, 2024-09-03:
Release 2.0.0 is now available, and it includes the fix that addresses this issue. You should no longer see these messages. I would greatly appreciate your feedback. If the issue is resolved to your satisfaction, please consider closing this thread.

Thank you!

m-unkel, 2024-09-23:
We have released a fix addressing this issue. Since we haven't heard back from you in a while, we assume the problem has been resolved. As a result, we are closing this issue.

If you're still experiencing any problems or have further questions, please feel free to reopen the issue or create a new one.
Thank you!

closed 2024-09-23 by m-unkel