Fraunhofer-FIT-DIEN/iec104-python #27: Problem when setting a boolean command

dkaranassos-s, 2024-09-09:
Hello, we are using the below simple code to change the value of command3 point, which has io_address 21 and type C_SC_NA_1. When trying to run it on Macbook the value doesn't change and we get a libc++abi: terminating due to uncaught exception of type pybind11::reference_cast_error: . When we are trying to run it on Windows the server side halts.

Do you have any insight?

[client.txt](https://github.com/user-attachments/files/16931892/client.txt)
[server.txt](https://github.com/user-attachments/files/16931893/server.txt)
[image]

m-unkel, 2024-09-09:
Hey, thanks for reporting this issue! 
Please allow me to ask for further details as the link to the screenshot is not accessible for me. 

Did you installed the binary version or build from source? Pypi is offering both options. 

What OS versions and Python versions are you using (Mac and Win)? 

I am not able to reproduce the problem yet, but I'll continue investigating.

dkaranassos-s, 2024-09-10:
Hello, no problem!
I installed the binary version, the OS versions are Sonoma 14.6.1 and Windows 10 Pro and Python3.11.2 and Python3.11 respectively. We are still facing the same issue so it wasn't a one time thing. I am re uploading the screenshot.
[image]

dkaranassos-s, 2024-09-10:
I mistakenly closed the issue. This still persists.

m-unkel, 2024-09-12:
Thank you for providing additional details.

I tested the library on an older Intel-based Mac with Python 3.11 and was unable to reproduce the issue. Here's the wheel I used for testing (unzip and run pip install filename.whl):
[c104-2.0.0-cp311-cp311-macosx_11_0_universal2.whl.zip](https://github.com/user-attachments/files/16974389/c104-2.0.0-cp311-cp311-macosx_11_0_universal2.whl.zip)

However, I assume you're using an ARM-based Mac? I will ask a colleague to test it on a newer Mac, but please note that macOS is not officially supported, even though the library is expected to work.

Are you using the latest version, 2.0.0?

Regarding Windows, I was also unable to reproduce the issue with the server hanging. One suggestion for the client would be to wait for the connection to be fully established before sending the command, or at least check the return value of the command transmission for success, as shown in the example below:
```python
import c104

client = c104.Client()
connection = client.add_connection(ip="127.0.0.1", port=2404, init=c104.Init.INTERROGATION)
station = connection.add_station(common_address=47)

command3 = station.add_point(io_address=21, type=c104.Type.C_SC_NA_1)
command3.value = True

client.start()

while connection.state != c104.ConnectionState.OPEN:
    print("Waiting for connection to {0}:{1}".format(connection.ip, connection.port))
    time.sleep(0.2)

if command3.transmit(cause=c104.Cot.ACTIVATION):
    print("-> SUCCESS")
else:
    print("-> FAILURE")

```

Also, note that the default constructor arguments for the client have changed to `tick_rate_ms=100` and `command_timeout_ms=100`. However, it is perfectly fine to omit them.

Based on the **client.txt** you provided, it is possible that the command is being sent before the connection is fully open, causing the transmission to fail. This failure will be indicated by a `False` return value.

dkaranassos-s, 2024-09-23:
We are using the same wheel.
1. Yes, it is an ARM-based Mac, thank you for checking even though it's not officially supported, though the same happened on a Windows Machine. 
2. Yes, this error came up with the latest version, we don't have the same issues with version 1.18.
3. I will try the code you send and check if it works properly in that case.
Thank you!

m-unkel, 2024-09-24:
After further investigation, I have been able to successfully reproduce the issue.

I am working on a fix and I will keep you updated on the progress.

Thank you again for your contribution to improving the project!

m-unkel, 2024-09-24:
Update:

The callback function on_step_command has to return a value of type c104.ResponseState as stated by the type hint in the signature.

If you add `return c104.ResponseState.SUCCESS` to the end of this function, the error is gone.
But i will try to improve the exception message.

m-unkel, 2024-09-24:
I pushed a fix to handle unexpected callback return values.
This fix will be included in the upcoming v2.0.1 release.
If you wish to, you can try out the improved error handling via installing from source branch:

`pip install git+https://github.com/Fraunhofer-FIT-DIEN/iec104-python.git@main`

m-unkel, 2024-09-25:
Release 2.0.1 is now available, and it includes the fix that addresses this issue. I would greatly appreciate your feedback. If the issue is resolved to your satisfaction, please consider closing this thread.

Thank you!

dkaranassos-s, 2024-10-03:
Thank you very much! We will check it again from our side and if anything comes up we will get back to you!

closed 2024-10-03 by dkaranassos-s