Fraunhofer-FIT-DIEN/iec104-python #29: Setting t0,t1,t2,t3, k,w parameters

AntonisMous, 2024-09-23:
Hello,

I would like to ask how is it possible to set the timing parameters t0,t1,t2,t3, k,w according to IEC 60870-5-104 using the c104.

m-unkel, 2024-09-23:
Hello AntonisMous,
Currently, these parameters are hard-coded.

For the server, the default values from lib60870-C are used.
For the client, it also relies on the defaults from lib60870-C, with the following exceptions for timing values:

- t0 = 2
- t1 = 1
- t2 = 2

We could make these parameters configurable in the future.

Could you please share more details about the scenario you're working on that requires modification of these values?

AntonisMous, 2024-09-23:
Hello,

Thank you for your immediate answer. 
I am working on a server where these variables should be configured

* t0 = 5s
* t1 = 7s
* t2 = 10s
* t3 = 20s
* k = 12 apdus
* w = 8 apdus

m-unkel, 2024-09-23:
The default values in lib60870-C (as used by the c104.Server) are actually quite close to what you're looking for:

- t0 = 10s (instead of 5s)
- t1 = 15s (instead of 7s)
- t2 = 10s
- t3 = 20s
- k = 12 APDUs
- w = 8 APDUs

Since we're dealing with timing in seconds and most messages are exchanged within milliseconds, I'm curious if you've encountered specific issues due to the lack of configuration for these parameters. Are you facing any noticeable connection problems or other problems because of this?

AntonisMous, 2024-09-23:
Thank you once again for you helpful and immediate answers.

At the moment, I am not facing noticeable connection problems. 

However, I would like to know if there is a possibility to modify this parameters, in case where the limitation of not being able to modify timing parameters become a barrier for my project.

Is a new version planned  which will include the modification of timing parameters?

m-unkel, 2024-09-23:
Not at the moment, but I will add your request as a feature for the upcoming version 2.1.0.

In the meantime, we are planning to release a hotfix (version 2.0.1) this week to address the #28 bug.

As for version 2.1.0, there isn't a set release date yet since your feature request is the first we've received since the release of version 2.0.0. However, I can assure you that we can implement the timing parameter modification within the next two month. So, unless you encounter urgent issues, your project should be safe moving forward.

m-unkel, 2024-10-05:
The requested feature is implemented. The **Connection** as well as the **Server** classes have a new `protocol_parameters` property. If you wish to, you can verify the new feature via installing from source branch:

```shell
pip install git+https://github.com/Fraunhofer-FIT-DIEN/iec104-python.git@main
```

The fix will be included in the upcoming release.

m-unkel, 2025-03-05:
Release 2.1.1 is now available, and it includes the feature that addresses this issue. I would greatly appreciate your feedback. If the issue is resolved to your satisfaction, please consider closing this thread.

Thank you!

closed 2025-04-16 by m-unkel