Fraunhofer-FIT-DIEN/iec104-python #32: Feature-Request: Allow to set the c104.type of a point

georgwerner-lemonbeat, 2024-10-18:
Hello,

we have a scenario where we might want to transmit a point in two different variations.

1. As Type 1 -> M_SP_NA_1 (Single Info without timestamp)
2. As Type 30 -> M_SP_TB_1  (Single Info with timestamp)

So basically the underlying data-type shall not change.
Depending on the scenario (Interrogation, Spontaneous) we would like to send the value either with or without timestamp.

Unfortunately there is no setter for the type of a point.

Is it possible to allow setting the type of an already changes point?

Thanks a lot in advance
Georg

georgwerner-lemonbeat, 2024-10-29:
Hello Martin,

thanks a lot for your help and solution proposals.

Fortunately we now agreed in our integration-prototype not mix message-types for already defined datapoints, thus following your design decision.

Switching the datatype of an already created entity would also allow  that you can witch from value-report to step-command. 
So I thinks it's good to stick to fixed types.

Answering your question: 
We don't have any challenges sending the timestamp too frequently.  The request to change the message type without having a timestamp for interrogation responses came from our test-partner (the reason for this is unclear).
I guess somewhow our partner also came to the conclusion that mixing types is not a good idea.

---------------

We also have been working on another topic, setting the daylight-saving-time bit for messages. 
I created a fork and implemented a prototype only for one message type.

This approach is not very generic and must be adapted for every message type.

Would great to have your feedback on this.

Shall I open another thread for this?

(Hint: We know that using the SU-BIT in iec104 according to the Din-Norm is not recommended).

Thanks in advance
Georg

m-unkel, 2024-11-07:
Hello Georg,
Thank you for your PR.

Regarding the timestamps, we can continue the discussion in this thread; next time, please feel free to open a parallel issue. I saw your pull request, but you closed it again. Generally, a contributor license agreement would be necessary for a pull request of this magnitude.

The timestamp in the protocol also comes with an invalid flag, which can be used for the previously mentioned marking of invalid timestamps instead of a defined "unset" state. There is the possibility of creating a wrapper class (i.e., `c104.DateTime`) that can be converted to native Python `datetime.datetime` objects and also provides access to the SU bit, IV bit, and SB bit. There are already functions for the bit flags in the lib60870-C, so encoding and decoding do not need to be programmed from scratch.

However, the question remains where the state of DST is stored. In your PR, the state was assigned to the server. But does it really make sense for state to vary per server/client instance of a running process? Otherwise, I would tend to consider a static class property as a global configuration that is valid for all servers and clients in the running Python process (c104.DateTime.summerTime = True).

Best
Martin

m-unkel, 2025-03-05:
The requested feature has been implemented. I am currently working on an example that demonstrates its use for internal clock synchronization via the clock sync command.

Since this implementation introduces breaking changes, I was unable to include it in the latest release—apologies for the slow progress. If you'd like, you can verify the new feature by installing it from the source branch.

```shell
pip install git+https://github.com/Fraunhofer-FIT-DIEN/iec104-python.git@features/datetime
```

This is an example on the different options a server could make use of the datetime feature:

```python
import c104
import time
from datetime import datetime, timezone, timedelta

def main():
    # server and station preparation
    server = c104.Server()

    station = server.add_station(common_address=47)
    station.auto_time_substituted = False

    point = station.add_point(io_address=11, type=c104.Type.M_ME_TD_1)

    # start
    server.start()

    while not server.has_active_connections:
        print("Waiting for connection")
        time.sleep(1)

    time.sleep(1)

    print("transmit point 0 (none)")
    dt0 = c104.DateTime(datetime.now(timezone.utc))
    dt0.daylight_saving_time = False
    dt0.invalid = True
    dt0.substituted = True
    point.info = c104.NormalizedInfo(
        c104.NormalizedFloat(0.34),
        c104.Quality(),
        dt0
    )
    point.transmit(c104.Cot.SPONTANEOUS)

    time.sleep(1)

    print("transmit point 1 (Offset)")
    dt1 = c104.DateTime(datetime.now(timezone(timedelta(seconds=3600))))
    dt1.daylight_saving_time = False
    print(dt1.value.isoformat())
    point.info = c104.NormalizedInfo(
        c104.NormalizedFloat(-0.34),
        c104.Quality(),
        dt1
    )
    point.transmit(c104.Cot.SPONTANEOUS)

    time.sleep(1)

    print("transmit point 2 (DST)")
    dt2 = c104.DateTime(datetime.now(timezone.utc))
    dt2.daylight_saving_time = True
    print(dt2.value.isoformat())
    point.info = c104.NormalizedInfo(
        c104.NormalizedFloat(0.35),
        c104.Quality(),
        dt2
    )
    point.transmit(c104.Cot.SPONTANEOUS)

    time.sleep(1)

    print("transmit point 3 (Station offset)")
    station.timezone_offset = 14400
    station.daylight_saving_time = True
    dt3 = c104.DateTime(datetime.now(timezone(timedelta(seconds=14400))))
    dt3.daylight_saving_time = False
    print(dt3.value.isoformat())
    point.info = c104.NormalizedInfo(
        c104.NormalizedFloat(-0.35),
        c104.Quality(),
        dt3
    )
    point.transmit(c104.Cot.SPONTANEOUS)

    time.sleep(1)

    print("transmit point 4 (Station offset+DST)")
    station.timezone_offset = 14400
    dt4 = c104.DateTime(datetime.now(timezone(timedelta(seconds=14400))))
    dt4.daylight_saving_time = True
    print(dt4.value.isoformat())
    point.info = c104.NormalizedInfo(
        c104.NormalizedFloat(0.36),
        c104.Quality(),
        dt4
    )
    point.transmit(c104.Cot.SPONTANEOUS)

    time.sleep(1)

    print("transmit point 5 (auto substituted)")
    point.value = c104.NormalizedFloat(0.0)
    point.transmit(c104.Cot.SPONTANEOUS)

    server.stop()

if __name__ == "__main__":
    print()
    print("START simple server")
    print()
    main()

```