Fraunhofer-FIT-DIEN/iec104-python #33: counter interrogation

Vaipishnick, 2024-10-23:
Hello!

How can I make server answer "True" at counter interrogation request from  client?

>>> import c104
>>> server = c104.Server(ip="0.0.0.0", port=2404)
>>> station = server.add_station(common_address=47)
>>> server.start()

>>> import c104
>>> client = c104.Client(tick_rate_ms=1000, command_timeout_ms=5000)
>>> connection = client.add_connection(ip="127.0.0.1", port=2404, init=c104.Init.NONE)  
>>> station = connection.add_station(common_address=47)
>>> client.start()

Then I tried to make request:
>>> connection.counter_interrogation(common_address=47, cause=c104.Cot.ACTIVATION, qualifier=c104.Qoi.STATION)
False

m-unkel, 2024-10-23:
Hello Vaipishnick,
you have to wait for a successful connection establishment before sending commands.
Starting the client does not wait for successful connection setups.
```python
    import time

    # ... your setup logic

    # start
    client.start()

    while connection.state != c104.ConnectionState.OPEN:
        print("Waiting for connection to {0}:{1}".format(connection.ip, connection.port))
        time.sleep(0.5)

    # ... now you can execute commands successfully
```

Looking forward to your feedback.
Best, 
Martin

m-unkel, 2025-03-05:
Since we haven't received any updates in a while, we are closing this issue.
If the problem persists or if you have any further questions, please feel free to open a new issue.

closed 2025-03-05 by m-unkel