Fraunhofer-FIT-DIEN/iec104-python #35: Can't change state of point with iot  M_EP_TD_1

Vaipishnick, 2024-10-29:
>>> import c104
>>> my_server = c104.Server(ip="0.0.0.0", port=2404, tick_rate_ms=1000, max_connections=10)
>>> station = my_server.add_station(common_address=34)
>>> point = station.add_point(io_address=3300, type=c104.Type.M_EP_TD_1)
>>> point.value
<EventState.OFF: 1>
>>> point.value = c104.EventState(0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid value, please provide an instance of the matching information value class (value.__class__)
>>> point.value = c104.EventState(2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid value, please provide an instance of the matching information value class (value.__class__)
>>> point.value = c104.EventState(3)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid value, please provide an instance of the matching information value class (value.__class__)

What is wrong?

m-unkel, 2024-11-04:
Dear Vaipishnick,

Thank you for bringing this issue to our attention. We can confirm that this problem is caused by pybind11 casting the value instance down to an integer due to its native integer compatibility, instead of correctly accepting the actual EventState object.

We are actively working on a hotfix to resolve this issue and aim to release the hotfix as soon as possible. I will keep you updated on our progress.

Thank you for your understanding and support.

m-unkel, 2024-11-04:
I wanted to inform you that we have just released a new update, version 2.0.2, which includes fixes and improvements.

Could you please try updating to the latest version and let us know if your issues have been resolved? Your input is invaluable to us.

Thanks again!

Vaipishnick, 2024-11-05:
[image]
All right. Thank you very much!

closed 2024-11-05 by m-unkel