Fraunhofer-FIT-DIEN/iec104-python #37: ModuleNotFoundError: No module named 'c104'

VuaeOF, 2024-11-15:
I'm trying to use c104==2.0.2 for openwrt with fixed python version: 3.7.

I compiled for mips and the package installed on the target system, but when importing it, I get an error:

```
import c104
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'c104'
```

I looked in the packages directory and only found this:

```
. ├── c104-2.0.2.dist-info
│   ├── INSTALLER
│   ├── LICENSE
│   ├── METADATA
│   ├── RECORD
│   ├── REQUESTED
│   ├── WHEEL
│   ├── direct_url.json
│   └── top_level.txt
├── c104.cpython-37-mips-linux-gnu.so
```

There is no python code, just metadata and a .so file. Is that how it should be?

m-unkel, 2024-11-15:
Hi VuaeOF,

Thanks for bringing this to our attention! You’re correct—the c104 folder, which contains the namespace with type hints, should be part of the source distribution. It seems to be missing in this instance.

As a temporary workaround, could you try installing the package directly from Git? You can do so with the following command:

```shell
pip install git+https://github.com/Fraunhofer-FIT-DIEN/iec104-python.git@fix/point-value-variant-backport
```

I’ll investigate why the folder isn’t being included in the source distribution, even though it’s listed in MANIFEST.in.

Thank you for your patience while I work on resolving this!

m-unkel, 2024-11-15:
After reviewing the issue further, I realized my initial response was too hasty. I apologize for the confusion.

The c104 folder with type hints is actually part of an upcoming feature and not included in version 2.0.2. For this version, the file system layout you’re seeing looks correct.

You can try copying the .so file into your working directory. This should allow you to import the module with: 'import c104' as well.

For all released versions the package only includes the .so file and the pip package’s metadata folder.

VuaeOF, 2024-11-18:
The problem was in the extension suffixes (I found this with -v -v).

Python on the target platform only looked for `['.cpython-37.so', '.abi3.so', '.so']` and the file name was c104.cpython-37-mipsel-linux-gnu.so

closed 2024-11-18 by VuaeOF