Fraunhofer-FIT-DIEN/iec104-python #38: Send a M_EI_NA_1 message

haroldnw, 2024-11-22:
Hello, thank you for the repo!

I've been trying to implement a server that needs to send a `M_EI_NA_1` message like so:
```
{  
    'commonAddress': 27808,
    'cot': <Cot.INITIALIZED: 4>,
    'elements': '00 00 00 00 ',
    'firstInformationObjectAddress': 0,
    'format': 'I',
    'negative': False,
    'numberOfObjects': 1,
    'originatorAddress': 0,
    'rx': 0,
    'sequence': False,
    'test': False,
    'tx': 0,
    'type': <Type.M_EI_NA_1: 70>
}
```

The code below raises the following error:

```
     point = station.add_point(io_address=0, type=c104.Type.M_EI_NA_1)
 ValueError: Unsupported type M_EI_NA_1
```

```python
def on_receive_raw_callback(server: c104.Server, data: bytes) -> None:
    data = c104.explain_bytes_dict(apdu=data)
    from pprint import pprint
    print("---------------------------------------------------------")
    pprint(data, indent=4)

server = c104.Server(
    ip="0.0.0.0", port=2404, tick_rate_ms=1000, max_connections=5
)
common_address = 27808

server.on_receive_raw(on_receive_raw_callback)
station = server.add_station(common_address=common_address)
server.start()

point = station.add_point(io_address=0, type=c104.Type.M_EI_NA_1)
point.transmit(cause=c104.Cot.INITIALIZED)
```

If I understand correctly, this message needs to be sent to the clients when the server has completed its initialization.
How would you implement this using this lib ? Would the `on_connect` callback be a good place to transmit this message ?

Thanks in advance!

haroldnw, 2024-12-11:
Thank you for your detailed answer! The proposed implementation idea seems good to me as it seems satisfy our needs.

When will this version **2.1** be released ? Looking forward to the server's `protocol_parameters` as well to change the `keep_alive_interval` property which seems to be set at 20000ms by default.

m-unkel, 2024-12-19:
Perfect! 
We planned to release version 2.1 in January. 

The value of 20 seconds for keep alive messages was defined by the underlying lib60870-C as default value. 
With version 2.1 you will be able to change that of course.

m-unkel, 2024-12-31:
The **End of Initialization** Support was added to the development version with commit ba54c0e.

Client side:
```python
def cl_on_station_initialized(client: c104.Client, station: c104.Station, cause: c104.Coi) -> None:
    print("STATION {0} INITIALIZED due to {1} | CLIENT OA {2}".format(station.common_address, cause, client.originator_address))

my_client.on_station_initialized(callable=cl_on_station_initialized)
```

Server side:
```
my_station = my_server.get_station(47)
my_station.signal_initialized(cause=c104.Coi.REMOTE_RESET)
```

We discussed internally if it makes sense to add the **signal_initialized** method to the **server** object or the **station** object, but since it is possible to reset a single station it is more flexibel to add this method the individual station object. This means if you would like to send the end of initialization message for multiple or all stations of a server, you need to loop over those stations and call the **signal_initialized** method for each station.

kevin-menon-NW-Groupe, 2025-01-03:
Thanks @m-unkel for your quick release! 
I'm implementing the initialization on our server at the moment. Howver do you have an idea when the release will be pushed to [pypi](https://pypi.org/project/c104/#history)? 
Thanks!

kevin-menon-NW-Groupe, 2025-01-03:
Also, just to ensure, when calling:
```python
my_station.signal_initialized(cause=c104.Coi.REMOTE_RESET)
```
the Cot of the payload ([Cot.INITIALIZED: 4](https://iec104-python.readthedocs.io/latest/python/enum/cot.html#c104.Cot.INITIALIZED)) will be handled by the CPP lib? 
In which case only the [Coi](https://iec104-python.readthedocs.io/latest/python/enum/coi.html#c104.Coi.LOCAL_POWER_ON) is provided by the python lib?

kevin-menon-NW-Groupe, 2025-01-06:
Many thanks!

m-unkel, 2025-03-05:
Release 2.1.1 is now available, and it includes the feature that addresses this issue. I would greatly appreciate your feedback. If the issue is resolved to your satisfaction, please consider closing this thread.

Thank you!