Fraunhofer-FIT-DIEN/iec104-python #39: Send command to an existing server

zxcvbasdf-clt, 2024-12-06:
Hi,
Thank you for your help with implementing the simulator in Python.

I am encountering an issue and would appreciate your guidance. I am using C104 to create a client and connect to a device that provides a 104 server. I can retrieve the monitor point from the server using C104, and I have attempted to modify the point value. According to the sample and documentation, I need to add the point with Control Direction on both the client and server sides. However, my server is not created using C104. Is there any way to achieve this?

Thank you in advance for your assistanc

m-unkel, 2024-12-06:
Hi,

We’re glad to hear you appreciate our simulator implementation! Regarding your question:

Unfortunately, we cannot assist directly with your server configuration if it is not a C104 server. If your server cannot be configured, it should ideally provide documentation that outlines the existing information object addresses and their associated type IDs.

In the unexpected scenario where your server expects the client to send commands on the same information object address that the server uses for monitoring events, please note that this is not supported by C104. This convention is in place to avoid potential type conflicts.

If you can share more details about your server and its requirements, we would be happy to provide additional suggestions or guidance where possible.

Thank you, and we look forward to assisting further!

m-unkel, 2025-03-05:
Since we haven't received any updates in a while, we are closing this issue.
If the problem persists or if you have any further questions, please feel free to open a new issue.