Fraunhofer-FIT-DIEN/iec104-python #40: Performing global station interrogations programmatically

Dosclic98, 2024-12-06:
I was interested in performing global station interrogation programmatically where the client asks the server to transmit all its measures. I've seen that during the initialization phase (in case the c104.Init.ALL or c104.Init.INTERROGATION  enums are used) where the client performs a general interrogation (ASDU type = C_IC_NA_1, IOA = 0 and QOI = 20) and the server responds with a series of I-type packets containing all the measures it exposes. You can find an exemple of whan I am refering to in the following image.

[image]

I was curious if it was possible to perform this kind of operation programmatically and not only during initialization. 
I want to reproduce the scenario where a SCADA control center asks an RTU to send all its measures periodically without it being configured to do so spontaneously. I know that an RTU should use periodic updates to do so but in reality it seems that some RTUs are not configured to perform periodic updates spontaneously but they are explicitly queried by the control center using general interrogations (so without having to send an interrogation for each specific data point separately).

lennart-bader, 2024-12-06:
The `Connection` class offers the respective methods for your use case.

With the connection object, you can call the `interrogation` and `counter_interrogation` methods. This works with the example server. Note that I explicitly disable the general interrogation when adding the connection.

```python
import c104
import time

def main():
    client = c104.Client()
    connection = client.add_connection(ip="127.0.0.1", port=2404, init= c104.Init.CLOCK_SYNC)
    station = connection.add_station(common_address=47)
    client.start()
    # Wait for the connection to be established
    while connection.state != c104.ConnectionState.OPEN:
            print("Waiting for connection to {0}:{1}".format(connection.ip, connection.port))
            time.sleep(1)
    # Send an interrogation command
    response = connection.interrogation(common_address=47, cause=c104.Cot.ACTIVATION, qualifier=c104.Qoi.STATION, wait_for_response=False)
    print(response)  # success

if __name__ == "__main__":
    main()

```

[image]

Dosclic98, 2024-12-06:
Thanks for the quick answer, that's exactly what I was looking for.
I asked this because I am trying to [extend](https://github.com/Dosclic98/wattson) your Wattson emulation testbed (btw that's a great tool) to be able to model this specific scenario that is usefull for my research.
Are you still actively working on the non public version of the framework?

lennart-bader, 2024-12-06:
Wattson is still under active development and I'm planning to publish a new release on Github soon as well.

closed 2024-12-06 by Dosclic98