Fraunhofer-FIT-DIEN/iec104-python #42: how to process when the main station sends "C_IC_NA_1"?

panshaoning, 2024-12-30:
How to handle the general call message from the main station? That is, how to process when the main station sends "C_IC_NA_1"?

m-unkel, 2024-12-30:
Hi panshaoning,
Thank you for your question!

The interrogation command (C_IC_NA_1) is handled automatically by the server instance in C104. When the main station sends a general call message, the server will:

1. Respond with an activation confirmation.
2. Send messages containing the current values of all monitoring points.
3. Conclude with an activation termination message.

This behavior ensures that the general interrogation is processed seamlessly without requiring additional implementation on your part.

If you have any further questions or encounter any issues, feel free to let us know!

Best regards,
Martin

panshaoning, 2024-12-31:
Thank you for your reply.

I have tested, and indeed, the instances of the C104 server can handle it automatically.

Below is the first message example of the total call (C_IC_NA_1) response from the c104 server:

`
  68-FA-02-00-02-00-0D-1E-14-00-2F-00-08-00-00-1B-49-7F-41-00-09-00-00-40-65-B9-42-00-0A-00-00-83-00-83-42-00-0B-00-00-E1-25-F4-40-00-0C-00-00-8F-0F-6E-42-00-0D-00-00-BE-34-76-41-00-0E-00-00-71-EF-49-42-00-0F-00-00-C6-71-AF-42-00-10-00-00-0E-FD-9D-41-00-11-00-00-42-56-12-42-00-12-00-00-D1-EF-89-41-00-13-00-00-E2-F7-84-42-00-14-00-00-B4-0F-BF-41-00-15-00-00-94-2E-47-42-00-16-00-00-91-B0-BE-42-00-17-00-00-B1-F2-78-42-00-18-00-00-9A-A2-A0-41-00-19-00-00-F4-6E-01-42-00-1A-00-00-94-71-12-42-00-1B-00-00-8E-EE-44-42-00-1C-00-00-9A-18-FD-41-00-1D-00-00-1B-B3-1A-41-00-1E-00-00-FE-2C-76-42-00-1F-00-00-4E-08-EE-40-00-20-00-00-2C-35-9A-42-00-21-00-00-38-9F-E0-40-00-22-00-00-A5-81-B5-41-00-23-00-00-BA-92-97-42-00-24-00-00-10-B6-B1-42-00-25-00-00-FC-EA-18-42-00`

This message contains 30 measurement points and data, but in our region, we only need to include 30 data  and do not need to include the measurement points.
What should I do?

m-unkel, 2024-12-31:
Regarding your question, I’d be happy to receive a desired example message to better understand how it should look optimally in your case. Do I understand correctly that your requirement is to send the message formatted as a sequence of data instead of explicitly including all information object addresses? Or are there other details that should be formatted differently?

Currently the library does not support messages formatted as sequence, but we will add support for it in the upcoming version 2.1.0 in January. Looking forward to your example!

panshaoning, 2024-12-31:
> Regarding your question, I’d be happy to receive a desired example message to better understand how it should look optimally in your case. Do I understand correctly that your requirement is to send [...]

Yes, the messages I want to send do not need to include object addresses. Because of the needs of my project, is it possible to handle this in sv_on_send_raw now?

m-unkel, 2024-12-31:
This is not currently supported in the _stable_ release.

The `on_send_raw` callback provides **read-only** access to analyze protocol bytes and does not allow for modifying any data.

I can suggest two options to address your requirement:

1. Build the latest development version of this repository:
I recently pushed a new commit that automatically attempts to use sequences for interrogation responses and periodic monitoring data if the IOAs are in order. However, please note that this version is not tested as stable, and we cannot provide official support for this development release.

   You can test it by installing the package directly from the source using the following command:
   ```shell
   pip install git+https://github.com/Fraunhofer-FIT-DIEN/iec104-python.git@main
   ```
   I would be happy to get some feedback on that. You need the build tools on your operating system for this installation setup.

2. Wait for the upcoming stable release:
The next release is expected within the next 30 days, although the exact date depends on available resources and any bugs identified during final testing. A release by the end of January is a realistic target.

panshaoning, 2025-01-01:
thank you.   I'll wait for the next release.

m-unkel, 2025-01-01:
The relevant code is part of the underlying lib60870-C and should be discussed there. You would need to patch the library. The library does not support a write access in this event handler.

Your best bet is my opinion is to try the dev version by installing from git.

panshaoning, 2025-01-04:
Thank you for your patient responses. I have currently written a slave station program using Python myself.

closed 2025-03-05 by m-unkel