Fraunhofer-FIT-DIEN/iec104-python #43: Incorrect parsing of the points from the first station

akiUp, 2025-01-09:
Type and IOA of first interrogated station gets rewrited  by previous value.
Here is the output from dump_client example, Station 2 supposed to have 4 points(SP_NA, ME_NA, ME_NB, ME_NC):

CL] |--+ CLIENT has 1 connections
       |--+ CONNECTION has 3 stations
          |--+ STATION 2 has 1 points
             |      TYPE      |   IOA   |       VALUE        |        PROCESSED AT        |        RECORDED  AT        |      QUALITY      
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
             | Type.M_SP_NA_1 |       1 |                0.0 | 2025-01-09T18:58:02.062122 |                      N. A. | Quality set: {}, is_good: True
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
          |--+ STATION 3 has 2 points
             |      TYPE      |   IOA   |       VALUE        |        PROCESSED AT        |        RECORDED  AT        |      QUALITY      
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
             | Type.M_BO_NA_1 |       1 | 0b00000000000000000000000000000000 | 2025-01-09T18:58:02.062231 |                      N. A. | Quality set: {}, is_good: True
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
             | Type.M_DP_NA_1 |       2 | Double.INTERMEDIATE | 2025-01-09T18:58:02.062845 |                      N. A. | Quality set: {}, is_good: True
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
          |--+ STATION 1 has 2 points
             |      TYPE      |   IOA   |       VALUE        |        PROCESSED AT        |        RECORDED  AT        |      QUALITY      
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
             | Type.M_ME_NA_1 |       1 |           0.499985 | 2025-01-09T18:58:02.063287 |                      N. A. | Quality set: {}, is_good: True
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
             | Type.M_SP_NA_1 |       2 |               True | 2025-01-09T18:58:02.066794 |                      N. A. | Quality set: {}, is_good: True
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------

m-unkel, 2025-01-09:
Hi,

Thank you for reporting this issue! To better understand and investigate the problem, could you please provide a minimal example code snippet that reproduces the issue? This will help us pinpoint the cause more effectively and provide a solution faster.

Looking forward to your response!

akiUp, 2025-01-09:
Hi this is just a dump_client.py script from examples section of this repository. https://github.com/Fraunhofer-FIT-DIEN/iec104-python/blob/main/examples/dump_client.py

Python version is 3.12.3 running on Ubuntu Desktop 24.04.1 LTS
Queries running against this simulator http://www.redisant.com/iec104server

m-unkel, 2025-01-10:
Thank you for sharing more details!

First I’ve tried to reproduce the issue using a c104.Server instead of the REDISANT simulator as the communication partner but was unable to observe the same behavior. Testing an unknown simulator in a secure environment may take some time.

To help us investigate further, it would be very helpful if you could provide the full log from the dump_client. Here’s how you can generate it:

1. Start your simulator.
2. Run the dump_client command and redirect all output to a file named output.txt:
    `python3 ./dump_client.py > ./output.txt 2>&1`
3. Stop your simulator.
4. Once the shell command exits, you can send the output.txt file to me, either here or via email.

Thank you for your cooperation!

akiUp, 2025-01-10:
> With the trial version, I’m unable to create more than one slave. However, I couldn’t identify any issues during my testing. Could you please double-check that your configuration is correct and ensure there are no IOA collisions on the same COA/Slave?
>
> This will help rule out configuration-related issues as a potential cause.

Hi Martin
attaching, the collected dump and slightly updated simulator configuration snapshot.
in this case first 2 values of station 3 got lost

Sent with [Proton Mail](https://proton.me/mail/home) secure email.

[c104.Client] Created
[c104.Client] set_originator_address] prev: 0 | new: 123
[c104.Client] add_connection] IP 192.168.117.200 | PORT 2404
[c104.Connection] Created
[c104.Connection] connect] Asynchronous connect to 192.168.117.200:2404
[c104.Connection] CALLBACK on_state_change
[c104.Connection] state] CLOSED -> CLOSED_AWAIT_OPEN
[c104.Client] start] Started
[c104.Connection] state] CLOSED_AWAIT_OPEN -> OPEN_MUTED
[c104.Connection] set_open] Opened connection to 192.168.117.200:2404
[c104.Connection] connection_handler] Connection OPENED to 192.168.117.200:2404 | TOTAL 120 µs
[c104.Connection] unmute] Unmute connection to 192.168.117.200:2404
[c104.Connection] raw_message_handler] Stats | TOTAL 2 µs
[c104.Connection] CALLBACK on_state_change
[c104.Connection] CALLBACK on_send_raw
[c104.Connection] raw_message_handler] Stats | TOTAL 1077 µs
[c104.Connection] CALLBACK on_receive_raw
[c104.Connection] set_muted] Unmuted connection to 192.168.117.200:2404
[c104.Connection] connection_handler] Connection ACTIVATED to 192.168.117.200:2404 | TOTAL 81 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 1 µs
[c104.Connection] await_command_success] Await 65535-C_IC_NA_1-0
[c104.Connection] raw_message_handler] Stats | TOTAL 16 µs
[c104.Connection] set_command_success] Result 65535-C_IC_NA_1-0: 1 | found: 1
[c104.Connection] asdu_handler] C_IC_NA_1 Response Stats | CA 65535 | IOA 0 | TOTAL 127 µs
[c104.Connection] await_command_success] Non-topic 65535-C_IC_NA_1-0
[c104.Connection] raw_message_handler] Stats | TOTAL 16 µs
[c104.Client] CALLBACK on_new_station
[c104.Connection] add_station] CA 1
[c104.Client] CALLBACK on_new_point
[c104.Connection] asdu_handler] M_SP_NA_1 Report Stats | CA 1 | IOA 1 | TOTAL 11516 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 1709 µs
[c104.Client] CALLBACK on_new_station
[c104.Connection] add_station] CA 2
[c104.Client] CALLBACK on_new_point
[c104.Connection] asdu_handler] M_BO_NA_1 Report Stats | CA 2 | IOA 1 | TOTAL 5494 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 9 µs
[c104.Client] CALLBACK on_new_point
[c104.Connection] asdu_handler] M_DP_NA_1 Report Stats | CA 2 | IOA 2 | TOTAL 3708 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 9 µs
[c104.Connection] asdu_handler] Invalid message format: Value is out of range.
[c104.Connection] raw_message_handler] Stats | TOTAL 1 µs
[c104.Client] CALLBACK on_new_station
[c104.Connection] add_station] CA 3
[c104.Client] CALLBACK on_new_point
[c104.Connection] asdu_handler] M_DP_NA_1 Report Stats | CA 3 | IOA 1 | TOTAL 3216 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 5 µs
[c104.Connection] asdu_handler] M_ME_NB_1 Report Stats | CA 3 | IOA 1 | TOTAL 1489 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 4 µs
[c104.Client] CALLBACK on_new_point
[c104.Connection] asdu_handler] M_ME_NC_1 Report Stats | CA 3 | IOA 2 | TOTAL 7835 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 5 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 15 µs
[c104.Connection] set_command_success] Result 65535-C_IC_NA_1-0: 1 | found: 1
[c104.Connection] asdu_handler] C_IC_NA_1 Response Stats | CA 65535 | IOA 0 | TOTAL 40 µs
[c104.Connection] await_command_success] Result 65535-C_IC_NA_1-0: 1
[c104.Connection] await_command_success] Stats 65535-C_IC_NA_1-0 | TOTAL -4908344 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 1 µs
[c104.Connection] await_command_success] Await 65535-C_CS_NA_1-0
[c104.Connection] raw_message_handler] Stats | TOTAL 15 µs
[c104.Connection] set_command_success] Result 65535-C_CS_NA_1-0: 1 | found: 1
[c104.Connection] asdu_handler] C_CS_NA_1 Response Stats | CA 1 | IOA 0 | TOTAL 78 µs
[c104.Connection] raw_message_handler] Stats | TOTAL 2 µs
[c104.Connection] set_command_success] Result 65535-C_CS_NA_1-0: 1 | found: 1
[c104.Connection] asdu_handler] C_CS_NA_1 Response Stats | CA 2 | IOA 0 | TOTAL 18 µs
[... 153 lines cut ...]
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
             | Type.M_ME_NC_1 |       2 | 1.6180000305175781 | 2025-01-10T21:16:40.441998 |                      N. A. | Quality set: {}, is_good: True
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------

CL] |--+ CLIENT has 1 connections
       |--+ CONNECTION has 3 stations
          |--+ STATION 1 has 1 points
             |      TYPE      |   IOA   |       VALUE        |        PROCESSED AT        |        RECORDED  AT        |      QUALITY      
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
             | Type.M_SP_NA_1 |       1 |               True | 2025-01-10T21:16:40.412715 |                      N. A. | Quality set: {}, is_good: True
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
          |--+ STATION 2 has 2 points
             |      TYPE      |   IOA   |       VALUE        |        PROCESSED AT        |        RECORDED  AT        |      QUALITY      
[... cut ...]
             | Type.M_DP_NA_1 |       1 |  3.140000104904175 | 2025-01-10T21:16:40.441733 |                      N. A. | Quality set: {}, is_good: True
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------
             | Type.M_ME_NC_1 |       2 | 1.6180000305175781 | 2025-01-10T21:16:40.441998 |                      N. A. | Quality set: {}, is_good: True
             |----------------|---------|--------------------|----------------------------|----------------------------|-------------------

m-unkel, 2025-01-11:
Thank you for sharing the log.

In the IEC 60870-5-104 protocol, Information Object Addresses (IOAs) must generally be unique within the same Common Object Address (COA) on a specific slave device. This means that you cannot have multiple distinct information objects (e.g., a M_SP_NA_1 and a M_ME_NA_1) sharing the same IOA (e.g., IOA = 1) under the same COA on the same slave.

From your log, I’ve extracted and commented on the relevant lines that demonstrate these conflicts:

### Conflict Examples from Log:

**Report COA=1, IOA=1**  
CL] <-in-- {'format': 'I', 'tx': 1, 'rx': 1, 'type': <Type.M_SP_NA_1: 1>, 'numberOfObjects': 1, 'commonAddress': 1, 'firstInformationObjectAddress': 1, 'elements': '01 00 00 01 '}  

**Report COA=2, IOA=1**  
CL] <-in-- {'format': 'I', 'tx': 2, 'rx': 1, 'type': <Type.M_BO_NA_1: 7>, 'numberOfObjects': 1, 'commonAddress': 2, 'firstInformationObjectAddress': 1, 'elements': '01 00 00 00 00 00 00 00 '}  

**Report COA=2, IOA=2**  
CL] <-in-- {'format': 'I', 'tx': 3, 'rx': 1, 'type': <Type.M_DP_NA_1: 3>, 'numberOfObjects': 1, 'commonAddress': 2, 'firstInformationObjectAddress': 2, 'elements': '02 00 00 03 '}  

**Report COA=3, IOA=1**  
CL] <-in-- {'format': 'I', 'tx': 4, 'rx': 1, 'type': <Type.M_ME_NA_1: 9>, 'numberOfObjects': 1, 'commonAddress': 3, 'firstInformationObjectAddress': 1, 'elements': '01 00 00 00 80 00 '}  

**Report COA=3, IOA=1**  
CL] <-in-- {'format': 'I', 'tx': 5, 'rx': 1, 'type': <Type.M_DP_NA_1: 3>, 'numberOfObjects': 1, 'commonAddress': 3, 'firstInformationObjectAddress': 1, 'elements': '01 00 00 02 '}  
-> **CONFLICT:** IOA 1 is already used at COA=3! Ignoring IOA 1.  

**Report COA=3, IOA=1**  
CL] <-in-- {'format': 'I', 'tx': 6, 'rx': 1, 'type': <Type.M_ME_NB_1: 11>, 'numberOfObjects': 1, 'commonAddress': 3, 'firstInformationObjectAddress': 1, 'elements': '01 00 00 7b 00 00 '}  
-> **CONFLICT:** IOA 1 is already used at COA=3! Ignoring IOA 1.  

**Report COA=3, IOA=1 and IOA=2**  
CL] <-in-- {'format': 'I', 'tx': 7, 'rx': 1, 'type': <Type.M_ME_NC_1: 13>, 'numberOfObjects': 2, 'commonAddress': 3, 'firstInformationObjectAddress': 1, 'elements': '01 00 00 c3 f5 48 40 00 02 00 00 a0 1a cf 3f 00 '}  
-> **CONFLICT:** IOA 1 is already used at COA=3! Ignoring IOA 1.  
-> IOA 2 is still accepted.

### Observations:
COA 3 has repeated conflicts where IOA 1 is reused across different types (M_ME_NA_1, M_DP_NA_1, M_ME_NB_1, etc.). This violates the uniqueness requirement for IOAs within a COA.
IOA 2 in COA 3 is accepted because it does not conflict with any previously received IOAs.

Best,
Martin

akiUp, 2025-01-12:
Hi,  Martin, I've rerun test with non interleaving IOA's and all fine.
There is no issue here, thank you for prompt response!

m-unkel, 2025-01-12:
Hi,
You’re welcome! I’m glad to hear that everything is working fine with non-interleaving IOAs.

If you like this project, we’d greatly appreciate it if you could give us a ⭐ on GitHub!

closed 2025-01-12 by m-unkel