Fraunhofer-FIT-DIEN/iec104-python #44: Building local wheel is not working (linux)

paul-clavier-nw, 2025-02-06:
Hello @m-unkel 

Thank you for the repository !
I am currently trying to build the wheel from the `main` branch (I want to have the last version so that this [commit](https://github.com/Fraunhofer-FIT-DIEN/iec104-python/commit/ba54c0ed218814d8f9acd1f13df5360b865f10f7) is included in my package :) )

I followed the steps describe in the [README](https://github.com/Fraunhofer-FIT-DIEN/iec104-python/blob/main/README.md#how-to-build-linux) and am using this Ubuntu distribution as whell as python `Python 3.10.12`, `cmake 3.22.1` and `ninja 1.10.1`
```
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy
```

At the step 3, when running `python3 -m pip wheel .` I get the following error (sorry for the long trace):

```sh
[21/42] Building CXX object CMakeFiles/_c104.dir/src/remote/Helper.cpp.o
      FAILED: CMakeFiles/_c104.dir/src/remote/Helper.cpp.o
      /usr/bin/c++ -DVERSION_INFO=\"2.1.0\" -D_c104_EXPORTS -I/home/paul/nwre/iec104-python/src -I/home/paul/nwre/iec104-python/depends/lib60870/lib60870-C/src/inc/api -I/home/paul/nwre/iec104-python/depends/lib60870/lib60870-C/src/hal/inc -I/home/paul/nwre/iec104-python/depends/lib60870/lib60870-C/src/common/inc -I/home/paul/nwre/iec104-python/depends/mbedtls/include -isystem /usr/include/python3.10 -isystem /home/paul/nwre/iec104-python/depends/pybind11/include -O3 -DNDEBUG -std=gnu++17 -fPIC -fvisibility=hidden -flto -fno-fat-lto-objects -Os -MD -MT CMakeFiles/_c104.dir/src/remote/Helper.cpp.o -MF CMakeFiles/_c104.dir/src/remote/Helper.cpp.o.d -o CMakeFiles/_c104.dir/src/remote/Helper.cpp.o -c /home/paul/nwre/iec104-python/src/remote/Helper.cpp
      In file included from /home/paul/nwre/iec104-python/src/types.h:58,
                       from /home/paul/nwre/iec104-python/src/module/ScopedGilAcquire.h:38,
                       from /home/paul/nwre/iec104-python/src/remote/Helper.cpp:33:
      /home/paul/nwre/iec104-python/src/enums.h:877:31: error: ‘MBEDTLS_TLS1_3_AES_128_GCM_SHA256’ was not declared in this scope; did you mean ‘MBEDTLS_TLS_PSK_WITH_AES_128_GCM_SHA256’?
        877 |   TLS1_3_AES_128_GCM_SHA256 = MBEDTLS_TLS1_3_AES_128_GCM_SHA256,
            |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |                               MBEDTLS_TLS_PSK_WITH_AES_128_GCM_SHA256
      /home/paul/nwre/iec104-python/src/enums.h:878:31: error: ‘MBEDTLS_TLS1_3_AES_256_GCM_SHA384’ was not declared in this scope; did you mean ‘MBEDTLS_TLS_PSK_WITH_AES_256_GCM_SHA384’?
        878 |   TLS1_3_AES_256_GCM_SHA384 = MBEDTLS_TLS1_3_AES_256_GCM_SHA384,
            |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |                               MBEDTLS_TLS_PSK_WITH_AES_256_GCM_SHA384
      /home/paul/nwre/iec104-python/src/enums.h:879:37: error: ‘MBEDTLS_TLS1_3_CHACHA20_POLY1305_SHA256’ was not declared in this scope; did you mean ‘MBEDTLS_TLS_PSK_WITH_CHACHA20_POLY1305_SHA256’?
        879 |   TLS1_3_CHACHA20_POLY1305_SHA256 = MBEDTLS_TLS1_3_CHACHA20_POLY1305_SHA256,
            |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |                                     MBEDTLS_TLS_PSK_WITH_CHACHA20_POLY1305_SHA256
      /home/paul/nwre/iec104-python/src/enums.h:880:31: error: ‘MBEDTLS_TLS1_3_AES_128_CCM_SHA256’ was not declared in this scope; did you mean ‘TLS1_3_AES_128_GCM_SHA256’?
        880 |   TLS1_3_AES_128_CCM_SHA256 = MBEDTLS_TLS1_3_AES_128_CCM_SHA256,
            |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |                               TLS1_3_AES_128_GCM_SHA256
      /home/paul/nwre/iec104-python/src/enums.h:881:33: error: ‘MBEDTLS_TLS1_3_AES_128_CCM_8_SHA256’ was not declared in this scope; did you mean ‘TLS1_3_AES_128_CCM_SHA256’?
        881 |   TLS1_3_AES_128_CCM_8_SHA256 = MBEDTLS_TLS1_3_AES_128_CCM_8_SHA256
            |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |                                 TLS1_3_AES_128_CCM_SHA256
      [22/42] Building CXX object CMakeFiles/_c104.dir/src/remote/message/PointCommand.cpp.o
      FAILED: CMakeFiles/_c104.dir/src/remote/message/PointCommand.cpp.o
      /usr/bin/c++ -DVERSION_INFO=\"2.1.0\" -D_c104_EXPORTS -I/home/paul/nwre/iec104-python/src -I/home/paul/nwre/iec104-python/depends/lib60870/lib60870-C/src/inc/api -I/home/paul/nwre/iec104-python/depends/lib60870/lib60870-C/src/hal/inc -I/home/paul/nwre/iec104-python/depends/lib60870/lib60870-C/src/common/inc -I/home/paul/nwre/iec104-python/depends/mbedtls/include -isystem /usr/include/python3.10 -isystem /home/paul/nwre/iec104-python/depends/pybind11/include -O3 -DNDEBUG -std=gnu++17 -fPIC -fvisibility=hidden -flto -fno-fat-lto-objects -Os -MD -MT CMakeFiles/_c104.dir/src/remote/message/PointCommand.cpp.o -MF CMakeFiles/_c104.dir/src/remote/message/PointCommand.cpp.o.d -o CMakeFiles/_c104.dir/src/remote/message/PointCommand.cpp.o -c /home/paul/nwre/iec104-python/src/remote/message/PointCommand.cpp
      In file included from /home/paul/nwre/iec104-python/src/types.h:58,
                       from /home/paul/nwre/iec104-python/src/module/ScopedGilRelease.h:38,
                       from /home/paul/nwre/iec104-python/src/module/GilAwareMutex.h:40,
                       from /home/paul/nwre/iec104-python/src/remote/message/IMessageInterface.h:35,
                       from /home/paul/nwre/iec104-python/src/remote/message/OutgoingMessage.h:35,
                       from /home/paul/nwre/iec104-python/src/remote/message/PointCommand.h:35,
                       from /home/paul/nwre/iec104-python/src/remote/message/PointCommand.cpp:32:
      /home/paul/nwre/iec104-python/src/enums.h:877:31: error: ‘MBEDTLS_TLS1_3_AES_128_GCM_SHA256’ was not declared in this scope; did you mean ‘MBEDTLS_TLS_RSA_WITH_AES_128_GCM_SHA256’?
        877 |   TLS1_3_AES_128_GCM_SHA256 = MBEDTLS_TLS1_3_AES_128_GCM_SHA256,
            |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |                               MBEDTLS_TLS_RSA_WITH_AES_128_GCM_SHA256
      /home/paul/nwre/iec104-python/src/enums.h:878:31: error: ‘MBEDTLS_TLS1_3_AES_256_GCM_SHA384’ was not declared in this scope; did you mean ‘MBEDTLS_TLS_PSK_WITH_AES_256_GCM_SHA384’?
        878 |   TLS1_3_AES_256_GCM_SHA384 = MBEDTLS_TLS1_3_AES_256_GCM_SHA384,
            |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |                               MBEDTLS_TLS_PSK_WITH_AES_256_GCM_SHA384
      /home/paul/nwre/iec104-python/src/enums.h:879:37: error: ‘MBEDTLS_TLS1_3_CHACHA20_POLY1305_SHA256’ was not declared in this scope; did you mean ‘MBEDTLS_TLS_PSK_WITH_CHACHA20_POLY1305_SHA256’?
        879 |   TLS1_3_CHACHA20_POLY1305_SHA256 = MBEDTLS_TLS1_3_CHACHA20_POLY1305_SHA256,
            |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |                                     MBEDTLS_TLS_PSK_WITH_CHACHA20_POLY1305_SHA256
      /home/paul/nwre/iec104-python/src/enums.h:880:31: error: ‘MBEDTLS_TLS1_3_AES_128_CCM_SHA256’ was not declared in this scope; did you mean ‘TLS1_3_AES_128_GCM_SHA256’?
        880 |   TLS1_3_AES_128_CCM_SHA256 = MBEDTLS_TLS1_3_AES_128_CCM_SHA256,
            |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            |                               TLS1_3_AES_128_GCM_SHA256
      /home/paul/nwre/iec104-python/src/enums.h:881:33: error: ‘MBEDTLS_TLS1_3_AES_128_CCM_8_SHA256’ was not declared in this scope; did you mean ‘TLS1_3_AES_128_CCM_SHA256’?
        881 |   TLS1_3_AES_128_CCM_8_SHA256 = MBEDTLS_TLS1_3_AES_128_CCM_8_SHA256
[... 451 lines cut ...]
        File "/tmp/pip-build-env-rf6wsycj/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/cmd.py", line 339, in run_command
          self.distribution.run_command(command)
        File "/tmp/pip-build-env-rf6wsycj/overlay/local/lib/python3.10/dist-packages/setuptools/dist.py", line 999, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-rf6wsycj/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 1002, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-rf6wsycj/overlay/local/lib/python3.10/dist-packages/setuptools/command/build_ext.py", line 99, in run
          _build_ext.run(self)
[... cut ...]
        File "<string>", line 122, in build_extension
        File "/usr/lib/python3.10/subprocess.py", line 526, in run
          raise CalledProcessError(retcode, process.args,
      subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--target', '_c104']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for c104
Failed to build c104
ERROR: Failed to build one or more wheels
```

m-unkel, 2025-02-11:
Hi @paul-clavier-nw ,
thank you for your interest in the repository!

The dependency on mbedtls was outdated, but this should now be fixed. Please try building again and let us know if the issue is resolved.

Best,
Martin

paul-clavier-nw, 2025-02-25:
Thank you very much @m-unkel it worked like a charm :)

Best,
Paul

closed 2025-03-05 by m-unkel