Fraunhofer-FIT-DIEN/iec104-python #45: Sending commands leads to "Segmentation Fault (core dumped)"

wiebsS, 2025-02-20:
Hi,

I'm trying to set up a testclient that connects to a station to receive data and send commands.
Reading a measurement value works great, but when I try to send a command I get "Segmentation fault (core dumped)".
There does not seem to be any other helpful debug messages.

The following is my testcode.
Any idea what could be the issue?
```
import c104
client = c104.Client()

connection = client.add_connection(ip="160.45.2.10", port=2404, init=c104.Init.ALL)
station = connection.add_station(common_address=1)

command_point = station.add_point(io_address=15, type=c104.Type.C_RC_TA_1)
command_point.command_mode = c104.CommandMode.SELECT_AND_EXECUTE

client.start()

while connection.state != c104.ConnectionState.OPEN:
    print("Waiting for connection to {0}:{1}".format(connection.ip, connection.port))
    time.sleep(1)

while True:
    if command_point.transmit(cause=c104.Cot.ACTIVATION):
        print("-> SUCCESS")
    else:
        print("-> FAILURE")
```

m-unkel, 2025-02-20:
Hi,

Thank you for reporting this issue! To better understand and diagnose the problem, could you provide the following details?

- Python version (python --version)
- c104 version
- Operating system and version
- Installation method (Did you install via pip, from source, or another method?)

Best,
Martin

wiebsS, 2025-02-20:
Hi, 
Thank you for the fast answer. I installed it via pip and it seems like the new version solved the issue.

Thanks a lot!

closed 2025-02-20 by wiebsS