Fraunhofer-FIT-DIEN/iec104-python #46: example "dump_client" recieve wrong values

unregistered33, 2025-03-01:
Thank you for your work! The "simple_client" example reads values from the server correctly, but when I started dump_client, I saw that only the first value was correct, and the other two were not. 

             |      TYPE                 |   IOA   |       VALUE                   |        PROCESSED AT                   |        RECORDED  AT        |      QUALITY      
             |---------------------------|--------|--------------------------------|-------------------------------------------|----------------------------|-------------------
             | Type.M_ME_NC_1 |       1 |             2017.0              | 2025-03-01T17:02:53.814322 |                      N. A.     | Quality set: {}, is_good: True
             |---------------------------|--------|--------------------------------|-------------------------------------------|----------------------------|-------------------
             | Type.M_ME_NC_1 |       2 |             **3147.0**              | 2025-03-01T17:02:53.955591 |                      N. A. | Quality set: {}, is_good: True
             |---------------------------|--------|--------------------------------|-------------------------------------------|----------------------------|-------------------
             | Type.M_ME_NC_1 |       3 |  **984.2926635742188** | 2025-03-01T17:02:53.955778 |                      N. A. | Quality set: {}, is_good: True
             |---------------------------|--------|--------------------------------|--------------------------------------------|----------------------------|-------------------
             | Type.M_ME_NC_1 |       4 |                0.0                  | 2025-03-01T17:02:53.251418 |                      N. A. | Quality set: {}, is_good: True
             |---------------------------|--------|--------------------------------|--------------------------------------------|----------------------------|-------------------

IO1 and IO4 is correct, but IO2 and IO3 is not. 
IO2  changes by decrement, IO3 by the EasyInSine function.
What could be the problem? Thanks!

m-unkel, 2025-03-05:
Thank you for your feedback and for reporting this issue!

To better understand the problem, could you please share the full output of the **dump_client** script? This will help us investigate further.

Additionally, could you provide details about the hardware and software you are using as the server? Are you running a c104.Server, or is it a different setup?

Looking forward to your response!

Best,
Martin

m-unkel, 2025-04-16:
Thank you for sharing the log.

In the IEC 60870-5-104 protocol, Information Object Addresses (IOAs) must generally be unique within the same Common Object Address (COA) on a specific slave device. This means that you cannot have multiple distinct information objects (e.g., a M_SP_NA_1 and a M_ME_NA_1) sharing the same IOA (e.g., IOA = 1) under the same COA on the same slave.

IOA 1 to 10 are reused across different groups or types (M_SP_TB_1, C_SC_NA_1, M_ME_NA_1) at the same station (1). This violates the uniqueness requirement for IOAs within a COA. Therefore only the first point per IOA will be accepted.

Best,
Martin

unregistered33, 2025-04-19:
Thank you, Martin!

closed 2025-04-19 by unregistered33