Fraunhofer-FIT-DIEN/iec104-python #47: Question about c104 point management

f0rw4rd, 2025-03-05:
I'm working with the c104 library to test IEC 60870-5-104 devices and have a question about point management.
When I try to test different command types for the same IOA, I find that after calling _station.add_point(io_address=X, type=C_SC_NA_1)_ a subsequent call to _station.add_point(io_address=X, type=C_DC_NA_1)_ throws an exception.
Is this the intended behavior of the API? Is there any way to remove a point from a station or update its type so I can test multiple command types on the same IOA?
Thanks for creating this excellent library - it's been incredibly helpful for my IEC 104 testing work!

m-unkel, 2025-03-06:
Yes, this is the expected behavior, as the current version does not support updating the type or removing a point after it has been added.

However, this is a sensible feature to have, and since the change was straightforward, I have already pushed the necessary updates to Git. A new release will be available soon, which will allow you to remove a point and recreate it with a different type.

Thank you for your feedback, and I'm glad to hear the library has been helpful!

m-unkel, 2025-03-06:
Release 2.2.0 is now available, and it includes the feature that addresses this issue. I would greatly appreciate your feedback! If the update resolves your issue, please consider giving the project a star and closing this thread.
Thank you! 😊

f0rw4rd, 2025-03-07:
Thank you for the quick implementation! I've tested release 2.2.0 and the point removal feature works perfectly. Being able to test different command types on the same IOA makes my workflow much smoother. Really appreciate your responsiveness and the excellent library - just gave it a star! ⭐

closed 2025-03-07 by f0rw4rd