Fraunhofer-FIT-DIEN/iec104-python #48: Server.protocol_parameters default values

ajeanNW, 2025-03-07:
Hello, thank you for the library that greatly facilitates communication via iec104 protocol.

We noticed an issue since the 2.1.1 release with `server.protocol_parameters`. The new release made this parameters editable with values in ms, yet when one creates a server instance the default values are still set to the old ones in seconds : t0 = 2, t1 = 2, t3 = 1 and t4 = 20.

With the default values we observed connection timeout loops, therefore we suggest to update those by a factor of 1000 to match the change of units.

m-unkel, 2025-03-07:
Thank you for reporting this issue!

The actual problem is with the documentation incorrectly stating that the values are in milliseconds—they are actually provided in seconds.

Even more concerning is the fact that you are experiencing timeout loops, as the timing values should not be too small. Have you been able to find settings that prevent these timeouts?

Could you share more details about your setup? This would help us reproduce and better understand the issue.
Looking forward to your feedback!

Best,
Martin

m-unkel, 2025-04-16:
Release 2.2.1 is now available, and it restores the original values as well as includes the fixed documentation. I would greatly appreciate your feedback. If the issue is resolved to your satisfaction, please consider closing this thread.

Thank you!

closed 2025-04-18 by ajeanNW