Fraunhofer-FIT-DIEN/iec104-python #49: Point.transmit() memory leak

ajeanNW, 2025-03-07:
Hello,

We use IEC104 protocol to send real-time data to a distant client for 485 different ioas : 
- We create a server instance and a station initialized with as many points. 
- Overall we send an average of 7000 messages per min.

We observe the following behavior : 
- The memory usage remains constant for the first hour.
- After an hour the memory usage starts increasing at a rate of 3 Mb per hour.
- When we disable the `point.transmit()` the memory usage stops increasing (we didn't initialize our points with a custom `on_before_transmit_callback`).

Do you have any idea of the source of this increase ? Is there a buffer on the server sending process that could explain this increase of memory usage ?

You'll find attached a screenshot of such memory usage behavior.

[image]

m-unkel, 2025-03-08:
Thank you for your detailed analysis.

I must admit that I haven’t tested with such a high number of points in a while, but there could be multiple potential sources for this issue. The underlying library, lib60870, is written in C and uses two queues per server with extensive raw pointer access. This suggests that the memory leak could originate there, but we need to investigate further.

It’s also possible that we missed freeing something on our end. However, since we rely on smart pointers, the only likely issue would be an interaction between C++ and Python reference counting and that should arise from the very beginning.

That said, resolving this may take some time. Due to limited resources and upcoming vacations, we won’t be able to provide deeper feedback before the end of April. We appreciate your patience and will update you as soon as we have more insights!

Best, 
Martin

m-unkel, 2025-03-10:
I performed an initial analysis using **Valgrind** and **GCC AddressSanitizer** but did not find any confirmed memory leaks. I also tested calling `transmit` for **700 points in a loop 10,000 times** without detecting any noticeable memory increase.  

That said, I did find a few **minor warnings** in Valgrind. These were unrelated to `transmit` and pointed to a **potential issue in the start and shutdown behavior**. These have already been addressed and fixed in Git.  

However, this does not entirely rule out a memory leak. The **pybind11 Python bindings** introduce multiple memory allocations that are only freed when the Python interpreter exits, but none of this memory appears to be permanently lost.  

To better understand the issue, could you provide a **sample script** of your server setup? Additionally, what is the reason for calling `transmit` instead of setting a `report_ms` interval on points? Using an interval would allow the library to **batch messages automatically**, which might help manage memory more efficiently. You could also manually handle batching using a **Batch object**.  

Or are you sending these points **only when their values change**? Understanding this would help us determine if there's a better approach to optimize performance.  

Looking forward to your response!

ajeanNW, 2025-03-12:
Thank you very much for the time you dedicated to helping us and finding the source of the issue. 

First of all, to give a bit more of context, we send points **both on a periodic and on a spontaneous basis**. We don't use an automatic transmit using `report_ms` for the later ones that must be sent **only on value change**. As for the first ones, our IEC104 protocol is at the end of a whole pipeline that involves multiple data aggregations and computations running periodically. On the IEC Server side, we kept the point transmission event-based (using transmit) firstly because **we prefer not to send any anything** when it happens that steps failed upstream rather than sending an old value that wasn't updated and secondly because we **don't always send the same points** depending on the some external business constraints.  

To get back on the issue we were encountering, we may have found the origin of the increasing memory usage that seems to come from **a callback waiting list rather than a memory leak**. We've explored this possibility since we encountered a deadlock error that wasn't catched by python :  

"""
terminate called after throwing an instance of 'std::runtime_error' 
  what():  Potential Deadlock: mutex Callback::callback_mutex waiting for lock > 100ms 
"""

We did actually used simple callbacks for the purpose of debug logging (`on_before_transmit_callback`, `on_receive_raw_callback`) and more complex ones (`on_receive_point_callback`) to handle incoming messages and feed them to our pipeline. But given the amount of messages we were transmitting and receiving (we receive an average of 200 messages per minutes besides the 7000 we send), they **seemed to be accumulating** and waiting an increasing amount of time for the lock to be released. 

 We removed debug callbacks (`on_before_transmit_callback`, `on_receive_raw_callback`) and implemented batch transmission and the memory usage of our server is now rather stable :  

[image]

m-unkel, 2025-03-16:
Thank you for sharing these details!

So the issue was more related to callback queuing rather than an actual memory leak. Given this, we might consider adding a warning message if the task queue grows beyond a certain threshold to indicate that callbacks are taking too long to execute.

If you enable debug mode for callbacks, it should log the execution time of each callback. Even with highly optimized callback code, the Python GIL enforces single-threaded execution, which can become a bottleneck. A 100ms deadlock detection is quite long in this context. Could you share more details on your callback implementation? This might help us identify potential optimizations.

Also, please be aware of bug #52 in batch messaging. If a batch exceeds the maximum packet size, not all data may be transmitted. As a temporary workaround, try limiting the number of points in one batch to 30.
The patched release is available in git but the release will not be available before mid of April.

Let us know if you need further assistance!

m-unkel, 2025-04-16:
Release 2.2.1 is now available, and it includes the fixes that addresses the batch transmission. I would greatly appreciate your feedback. If the issue is resolved to your satisfaction, please consider closing this thread.

Thank you!

closed 2025-04-18 by ajeanNW