Fraunhofer-FIT-DIEN/iec104-python #50: Send Control from C104 client to other server

sahas7997, 2025-03-07:
Hi,

Firstly, glad to see this in python version :) Much appreciated..!!

We are trying to use this only for a client simulation. We would like to use this opensource client simulator to connect to an 104 server device. However, we observed that c104 client is unable to decode the ActCon (may be) and hence getting 'False' response for transmit command of Control IOA. Below is the code snippet for easy reference :

client = c104.Client()
connection = client.add_connection(ip="127.0.0.1", port=2404, init=c104.Init.ALL)
connection.on_unexpected_message(callable=con_on_unexpected_message)
station = connection.add_station(common_address=47)

command = station.add_point(io_address=9001, type=c104.Type.C_SC_NA_1)
command.value = True

print("command.transmit(cause=c104.Cot.ACTIVATION)")

Also, is there any way to send SELECT and EXECUTE commands separately ?

Regards,
Satish

m-unkel, 2025-03-08:
Thank you for your feedback! Glad to hear you’re finding the Python version useful.

By default, the module operates in direct command mode. If you’d like to use select and execute mode, you can configure it when adding the point:

```python
command = station.add_point(io_address=9001, type=c104.Type.C_SC_NA_1, command_mode=c104.CommandMode.SELECT_AND_EXECUTE)
```

However, manually sending SELECT and EXECUTE commands separately is not supported.

Let us know if you have any further questions!
Best, 
Martin

sahas7997, 2025-03-08:
Hi Martin, 

Thanks for your quick response. 

The issue we are facing is,
c104 client is able to send Controls (DIRECT OR SELECT_AND_EXECUTE)
The control is being received and processed on server device **_(not c104)_**
On wireshark, we observe that ACTIVATION being received by 104 server device and ACTCON and ACTTERM is being sent from server to c104 client. 
However, we are getting response as False from transmit command.

command = station.add_point(io_address=9001, type=c104.Type.C_SC_NA_1, command_mode=c104.CommandMode.DIRECT)
response = command.transmit(cause=c104.Cot.ACTIVATION)

Can you please let us know if we are missing anything here ?

Regards,
Satish

m-unkel, 2025-03-08:
Thank you for the details!

The False response from transmit is likely due to a timeout that starts when the command is sent. Here are a few things to check:

1. Ensure the connection is fully established before transmitting:
If the connection isn't open yet, the command may time out before the response arrives. You can wait for the connection to open like this:
```python
while connection.state != c104.ConnectionState.OPEN:
    print("Waiting for connection to {0}:{1}".format(connection.ip, connection.port))
    time.sleep(1)
```

2. Adjust the command timeout:
The timeout can be configured when creating the client. Try increasing it to see if it helps:
```python
my_client = c104.Client(command_timeout_ms=1000)
```

3. Enable debug output to gather more details:
If the issue persists, please enable debug logging and share the output. This will help us determine whether the timeout is indeed the cause:
```python
c104.set_debug_mode(mode=c104.Debug.Client | c104.Debug.Connection)
```

Best
Martin

sahas7997, 2025-03-10:
Thanks for your inputs, Martin.!!

It worked after changing the timeout 🙂

I would like to ask if there is any way to extract the COT (like ActCon, ActTerm, negative responses - unknown IOA) from Control response. It will be helpful for the T104 client user to get more info about the Control execution.

Thanks, 
Satish

m-unkel, 2025-03-11:
If you want to provide the user with information about what happens at the protocol level, you can use hooks to capture and print relevant messages. For example:

```python
def cl_ct_on_receive_raw(connection: c104.Connection, data: bytes) -> None:
    print("OUT-->| {1} [{0}] | CONN OA {2}".format(data.hex(), c104.explain_bytes_dict(apdu=data), connection.originator_address))

def cl_ct_on_send_raw(connection: c104.Connection, data: bytes) -> None:
    print("IN <--| {1} [{0}] | CONN OA {2}".format(data.hex(), c104.explain_bytes_dict(apdu=data), connection.originator_address))

def cl_on_unexpected_message(connection: c104.Connection, message: c104.IncomingMessage, cause: c104.Umc) -> None:
    print("IN <--| Rejected Message | {1} from STATION CA {0}:{3}".format(message.common_address, cause))

cl_connection_1.on_receive_raw(callable=cl_ct_on_receive_raw)
cl_connection_1.on_send_raw(callable=cl_ct_on_send_raw)
cl_connection_1.on_unexpected_message(callable=cl_on_unexpected_message)
```

This allows you to print the Cause of Transmission (COT), including ActCon, ActTerm, or negative responses such as unknown IOA, and display them to the user.

However, there is no interactive select and Execute call, as the module is designed to handle this protocol flow as efficiently as possible. 

Best
Martin

Satish7997, 2025-03-14:
Thanks for all the details, Martin. Your inputs helped..!!
You can close this issue 👍

sahas7997, 2025-03-14:
Thanks for all the details, Martin. Your inputs helped..!!
Closing this issue 👍

closed 2025-03-14 by sahas7997