Fraunhofer-FIT-DIEN/iec104-python #51: Connection Auto-Disconnection

nobu121, 2025-03-13:
When running the batch_server example, the client connection automatically disconnects after a few seconds. This issue occurs with both the simulator client and QTester104, resulting in a socketError.

### Steps to Reproduce
1. Start the batch_server example
2. Connect using either simulator client or QTester104
3. After a few seconds, the connection automatically drops with socketError

```
START batch server

[c104.Server] Created
[c104.Server] add_station] CA 47
add point 0
add point 1
....
add point 29
[c104.Message] Created (outgoing) at 2168884535312
[c104.Message] Batch Created SPONTANEOUS at 2168884535312
[c104.Message] Point added to batch 43
[c104.Message] Point added to batch 44
[c104.Message] Point added to batch 45
<c104.Batch common_address=47, type=M_ME_TE_1, cot=SPONTANEOUS, number_of_objects=3, test=False, negative=False, sequence=True at 0x1f8fb97a010>
[c104.Server] start] Started
Waiting for connection
Waiting for connection
Waiting for connection
Waiting for connection
[c104.Server] connection_event_handler] Connection OPENED by 127.0.0.1:58233 | TOTAL 10 Âµs
[c104.Server] connection_event_handler] Connection ACTIVATED by 127.0.0.1:58233 | TOTAL 1 Âµs
[c104.Message] Created (incoming)
[c104.Server] send_activation_confirmation] to all MTUs
[c104.Message] Created (outgoing) at 2168889411904
[c104.Message] Batch Created REQUESTED_BY_GENERAL_COUNTER at 2168889411904
[c104.Message] Point added to batch 8
[c104.Message] Point added to batch 9
.....
[c104.Message] Point added to batch 37
[c104.Message] Created (outgoing) at 2168889412400
[c104.Message] Batch Created REQUESTED_BY_GENERAL_COUNTER at 2168889412400
[c104.Message] Point added to batch 40
[c104.Message] Point added to batch 43
[c104.Message] Point added to batch 44
[c104.Message] Point added to batch 45
Type.M_ME_NC_1 BEFORE READ or INTERROGATION on IOA: 8 VALUE: 60.64217758178711
[c104.Message] Created (outgoing) at 2168890278976
[c104.Message] Removed (outgoing) at 2168890278976
Type.M_ME_NC_1 BEFORE READ or INTERROGATION on IOA: 9 VALUE: 12.108438491821289
[c104.Message] Created (outgoing) at 2168890280256
[c104.Message] Removed (outgoing) at 2168890280256
.....
Type.M_ME_NC_1 BEFORE READ or INTERROGATION on IOA: 36 VALUE: 11.415517807006836
[c104.Message] Created (outgoing) at 2168890280256
[c104.Message] Removed (outgoing) at 2168890280256
Type.M_ME_NC_1 BEFORE READ or INTERROGATION on IOA: 37 VALUE: 31.74464225769043
[c104.Message] Created (outgoing) at 2168890283328
[c104.Message] Removed (outgoing) at 2168890283328
[c104.Server] send] Send Batch M_ME_NC_1 | COT: REQUESTED_BY_GENERAL_COUNTER | TOTAL 6814 Âµs
[c104.Message] Batch Removed REQUESTED_BY_GENERAL_COUNTER at 2168889411904
[c104.Message] Removed (outgoing) at 2168889411904
[c104.Message] Created (outgoing) at 2168890284096
[c104.Message] Removed (outgoing) at 2168890284096
[c104.Message] Created (outgoing) at 2168890285376
[c104.Message] Removed (outgoing) at 2168890285376
[c104.Message] Created (outgoing) at 2168890284096
[c104.Message] Removed (outgoing) at 2168890284096
[c104.Message] Created (outgoing) at 2168890281280
[c104.Message] Removed (outgoing) at 2168890281280
[c104.Server] send] Send Batch M_ME_TE_1 | COT: REQUESTED_BY_GENERAL_COUNTER | TOTAL 459 Âµs
[c104.Message] Batch Removed REQUESTED_BY_GENERAL_COUNTER at 2168889412400
[c104.Message] Removed (outgoing) at 2168889412400
[c104.Message] Removed (incoming)
[c104.Server] interrogation_handler] | IP 127.0.0.1:58233 | OA 0 | CA 47 | TOTAL 10012 Âµs
transmit batch 1
[c104.Message] Created (outgoing) at 2168890285376
[c104.Message] Removed (outgoing) at 2168890285376
[c104.Message] Created (outgoing) at 2168890280256
[c104.Message] Removed (outgoing) at 2168890280256
[c104.Message] Created (outgoing) at 2168890278720
[c104.Message] Removed (outgoing) at 2168890278720
[c104.Server] send] Send Batch M_ME_TE_1 | COT: SPONTANEOUS | TOTAL 504 Âµs
[c104.Server] connection_event_handler] Connection CLOSED by 127.0.0.1:58233 | TOTAL 19 Âµs
transmit batch 2

````

[image]

### Environment
- Library version: 2.2.0, 2.1.0
- Previous working version: 2.0.2
- OS: Windows
- Python version: 3.10

Could you please investigate what changes in the newer versions might be causing this connection stability issue?

m-unkel, 2025-03-15:
Thank you for reporting this issue!

This may be related to the changed protocol timing parameters in recent versions. Please try adjusting them manually to see if that improves connection stability.

We plan to increase the default timing parameters in the next release to prevent unexpected disconnections.

```python

connection.protocol_parameters.connection_timeout = 20
connection.protocol_parameters.message_timeout = 20
connection.protocol_parameters.confirm_interval = 10
connection.protocol_parameters.keep_alive_interval = 20

server.protocol_parameters.connection_timeout = 20
server.protocol_parameters.message_timeout = 20
server.protocol_parameters.confirm_interval = 10
server.protocol_parameters.keep_alive_interval = 20
```

Let us know if adjusting the timing helps or if you need further assistance!

nobu121, 2025-03-17:
> Thank you for reporting this issue!
> 
> This may be related to the changed protocol timing parameters in recent versions. Please try adjusting them manually to see if that improves connection stability.
> 
> We plan to increase the default timing parameters in the next release to prevent unexpected disconnections.
> 
> connection.protocol_parameters.connection_timeout = 20
> connection.protocol_parameters.message_timeout = 20
> connection.protocol_parameters.confirm_interval = 10
> connection.protocol_parameters.keep_alive_interval = 20
> 
> server.protocol_parameters.connection_timeout = 20
> server.protocol_parameters.message_timeout = 20
> server.protocol_parameters.confirm_interval = 10
> server.protocol_parameters.keep_alive_interval = 20
> Let us know if adjusting the timing helps or if you need further assistance!

Thank you for your prompt response and solution! 

I've implemented the suggested timing parameter adjustments in my server code, and I'm happy to report that the connection stability has significantly improved. We haven't experienced any unexpected disconnections since making these changes.

Your detailed explanation and the specific parameter values were very helpful. Thanks again for your assistance and for planning to address this in the next release.

Best regards

closed 2025-03-17 by nobu121