Fraunhofer-FIT-DIEN/iec104-python #52: Automatic batch creation fails at 30+ data points

lennart-bader, 2025-03-13:
When adding more than 30 data points with the same type with an automatic report interval on the server, only 30 points are transmitted during the periodic reports and during the general interrogation.

I.e., when transmitting more than 30 data points with the same type, only the first batch packet is sent.

This issue exists with `c104==2.2.0` but does not occur with `2.0.1`.

Attached you can find two PCAPs with  `c104==2.0.1` vs `c104==2.2.0` on the server side (client uses `2.2.0` in both cases).  
I also highlight the difference in the following screenshots from Wireshark:

Version 2.0.1:  
[image]

Version 2.2.0 (The second M_ME_NC_1 packet is missing):  
[image]

[c104-batching-PCAPs.tar.gz](https://github.com/user-attachments/files/19226899/c104-batching-v2.tar.gz)

m-unkel, 2025-03-14:
Thanks for reporting this issue. 

Could you please verify this is fixed in the latest main branch?

lennart-bader, 2025-03-20:
Thank you for the fix.  
I just installed the latest master version and the batching issue seems to be fixed, indeed.

**However**, this only applies to the batches during the general interrogation which now cover all data points.  
Periodic transmissions are not sent at all. I find periodic log statements for `on_receive_raw` and `on_send_raw` which match with `TESTFR act` and `TESTFR con`. No additional IEC 104 traffic is recorded (with Wireshark).

Manually reading a data point still works. There are no errors in the log:

```
Starting RtuDeployment
2025-03-20 09:22:29,420 - RTU 415 - INFO - Starting RTU 415
2025-03-20 09:22:29,420 - RTU 415 - INFO - Primary IP: 172.16.2.2
2025-03-20 09:22:29,424 - RTU 415 - INFO - Creating Wattson Client
2025-03-20 09:22:29,424 - RTU 415.SpontaneousLogic - INFO - Instantiating RTU Logic
2025-03-20 09:22:29,425 - RTU 415.IEC104 - INFO - Initialized RtuIec104
2025-03-20 09:22:29,425 - RTU 415.IEC104 - INFO - Adding Server Socket: 172.16.2.2:2404
[c104.Server] Created
[c104.Server] add_station] CA 415
[c104.Point] Created
[c104.Point] Created
[c104.Point] Created
[c104.Point] Created
[c104.Point] Created
[c104.Point] Created
2025-03-20 09:22:29,429 - RTU 415.DataPointManager - INFO - Initialized DataPointManager on host 172.16.2.2
2025-03-20 09:22:29,444 - PublishClient - INFO - Connecting to tcp://10.0.0.1:51001
2025-03-20 09:22:29,446 - WattsonClient - INFO - Connecting to tcp://10.0.0.1:51000
2025-03-20 09:22:29,465 - WattsonClient - INFO - Registered as n415_1
2025-03-20 09:22:29,465 - RTU 415 - INFO - Starting RTU 415
2025-03-20 09:22:29,465 - RTU 415.DataPointManager - INFO - Initializing Provider of type POWER_GRID
2025-03-20 09:22:29,578 - PowerGridProvider - INFO - Got Simulation start time: 2025-03-20T08:22:20.819093+00:00 // 2025-03-20T09:22:20.819093+01:00
2025-03-20 09:22:29,578 - RTU 415 - INFO - RTU Data Point Manager ready, starting Protocol Sockets
2025-03-20 09:22:29,578 - RTU 415 - INFO - Starting protocol sockets
[c104.Server] start] Started
2025-03-20 09:22:29,594 - RTU 415.SpontaneousLogic - INFO - set()
[c104.Server] CALLBACK on_connect
2025-03-20 09:22:42,897 - RTU 415.IEC104 - INFO - Client 172.16.1.2 attempting to connect. allowed=True (True)
[c104.Server] connection_event_handler] Connection OPENED by 172.16.1.2:44638 | TOTAL 9 µs
[c104.Server] connection_event_handler] Connection ACTIVATED by 172.16.1.2:44638 | TOTAL 1 µs
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] send_activation_confirmation] to all MTUs
[c104.Server] CALLBACK on_receive_raw
[c104.Point] CALLBACK on_before_read at IOA 10010
2025-03-20 09:22:42,925 - RTU 415.IEC104 - INFO - Received C_IC_NA_1 (General Interrogation)
[c104.Server] CALLBACK on_send_raw
[c104.Point] CALLBACK on_before_read at IOA 10020
[c104.Point] CALLBACK on_before_read at IOA 10030
[c104.Point] CALLBACK on_before_read at IOA 10040
[c104.Point] CALLBACK on_before_read at IOA 10050
[c104.Point] CALLBACK on_before_read at IOA 10060
[c104.Server] CALLBACK on_send_raw
[c104.Server] send] Send Batch M_ME_NC_1 | COT: REQUESTED_BY_GENERAL_COUNTER | TOTAL 948 µs
[c104.Server] interrogation_handler] | IP 172.16.1.2:44638 | OA 0 | CA 415 | TOTAL 1009 µs
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_clock_sync
[c104.Server] CALLBACK on_receive_raw
[c104.Server] clock_sync_handler] TIME 2025-03-20T09:22:42.942000+0100
[c104.Server] send_activation_confirmation] to all MTUs
[c104.Server] asdu_handler] TYPE C_CS_NA_1 | IP 172.16.1.2:44638 | OA 0 | CA 415 | TOTAL 122 µs
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Point] CALLBACK on_before_read at IOA 10030
[c104.Server] CALLBACK on_receive_raw
[c104.Server] send] Send M_ME_NC_1 | COT: REQUEST | TOTAL 7 µs
[c104.Server] send_activation_confirmation] to requesting MTU
[c104.Server] read_handler] IOA 10030 | IP 172.16.1.2:44638 | OA 0 | CA 415 | TOTAL 281 µs
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
[c104.Server] CALLBACK on_receive_raw
[c104.Server] CALLBACK on_send_raw
```

m-unkel, 2025-04-11:
The latest commit contains multiple fixes related to period transmission as well as interrogation handling.
Could you please rerun your test against the latest main branch?

lennart-bader, 2025-04-12:
Thank you for the fix!

Indeed, more than 30 data points are now transmitted during the general interrogation and periodically.  
In my example, I have 42 short values on a single station which are divided on a 30 IO-packet and a 12 IO-packet.

[image]

closed 2025-04-12 by lennart-bader