Fraunhofer-FIT-DIEN/iec104-python #54: Question about General Interrogation handling with Global Common Address

nobu121, 2025-03-18:
I noticed in the code that when handling General Interrogation with Global Common Address 65535(0xFFFF), the activation confirmation is sent multiple times - once for each station:
```cpp
void Server::sendActivationConfirmation(IMasterConnection connection,
                                        CS101_ASDU asdu, bool negative) {
  if (!isExistingConnection(connection))
    return;

  CS101_ASDU_setCOT(asdu, CS101_COT_ACTIVATION_CON);
  CS101_ASDU_setNegative(asdu, negative);

  if (isGlobalCommonAddress(CS101_ASDU_getCA(asdu))) {
    DEBUG_PRINT(Debug::Server, "send_activation_confirmation] to all MTUs");
    std::lock_guard<Module::GilAwareMutex> const st_lock(station_mutex);
    for (auto &s : stations) {
      CS101_ASDU_setCA(asdu, s->getCommonAddress());
      IMasterConnection_sendASDU(connection, asdu);
    }
  } else {
    DEBUG_PRINT(Debug::Server,
                "send_activation_confirmation] to requesting MTU");
    IMasterConnection_sendASDU(connection, asdu);
  }
}
```
According to my understanding of IEC 60870-5-104, when receiving a General Interrogation command with Global Common Address:

[image]
I believe the General Interrogation sequence should follow the pattern shown in the image:
1. One single ACTIVATION_CON message in response to the interrogation command
2. When COA=65535 (Global Common Address), data points from all stations should be returned with COT=20 (INTERROGATED_BY_STATION)
3. Finally, one ACTIVATION_TERMINATION message to conclude the sequence

This implementation ensures a clean and protocol-compliant message flow while properly handling global interrogation requests.

Could you explain the rationale behind sending multiple activation confirmations? Is this a specific requirement from the protocol that I'm missing?
Thank you for any clarification on this implementation approach.

lennart-bader, 2025-03-25:
The individual responses with C_IC ACTIVATION_CON are according to the standard.  

An interrogation triggered by the client with the global common address (`0xFFFF`) should trigger individual responses from each logical station. These responses do not have to be identical.

I.e., with two stations receiving the interrogation command, one might be unavailable and respond with a negative confirmation while the other one could still respond with a positive confirmation.

This is also included in `DIN EN 60870-5-101` in Figure 91.

So at least from my understanding, the current implementation is correct and  https://github.com/Fraunhofer-FIT-DIEN/iec104-python/pull/53 is not in accordance with the standard.

nobu121, 2025-03-26:
> The individual responses with C_IC ACTIVATION_CON are according to the standard.
> 
> An interrogation triggered by the client with the global common address (`0xFFFF`) should trigger individual responses from each logical station. These responses do not have to be identical.
> 
> I.e., with two stations receiving the interrogation command, one might be unavailable and respond with a negative confirmation while the other one could still respond with a positive confirmation.
> 
> This is also included in `DIN EN 60870-5-101` in Figure 91. Edit: And also part of the textual description Section 7.4.5: (Roughly translated from German): If an Interrogation Command is sent using the Common Address `FFFF`, then the all Logical Remote Units answer with their respective Common Address - in particular, they answer in exactly the same way as if they were interrogated by their respective individual common address.
> 
> So at least from my understanding, the current implementation is correct since the individual stations should not differentiate between an interrogation command with the global COA and their individual COA.

Thank you for clarifying the standard requirements. Based on your explanation, I now understand that the current implementation correctly handles activation confirmations for global interrogations (0xFFFF), where each station responds independently with its own common address. 

But In our recent tests, we observed an inconsistency, two stations sent activation confirmations (ACT-CON),only one station subsequently returned the requested data via a general interrogation response (C_IC).

Could you please clarify if this behavior is also compliant with the standard?

nobu121, 2025-03-27:
> Thanks for getting back at this!
> 
> This seems to be incorrect behavior according to the standard. Which version of c104 do you use? Could you share your code that results in this behavior?
> 
> I conducted a short test with a single server with two stations. Here, both stations correctly transmit their data after the interrogation request. I test it with `c104==2.0.1` since the current release has an issue.
> 
> [image]
> 
> For the client, the following events occur:
> 
> ```
> CL] Waiting for connection to 127.0.0.1:2404
> CL] Connection State Changed 0 | State ConnectionState.CLOSED_AWAIT_OPEN
> 
> CL] Connection State Changed 0 | State ConnectionState.OPEN_MUTED
> 
> CL] Connection State Changed 0 | State ConnectionState.OPEN
> 
> CL] NEW STATION 200 | CLIENT OA 0
> 
> CL] NEW POINT: Type.M_ME_NC_1 with COA 200 IOA 10 | CLIENT OA 0
> 
> CL] NEW POINT: Type.M_ME_NC_1 with COA 200 IOA 11 | CLIENT OA 0
> 
> CL] NEW STATION 300 | CLIENT OA 0
> 
> CL] NEW POINT: Type.M_ME_NC_1 with COA 300 IOA 11 | CLIENT OA 0
> 
> CL] NEW POINT: Type.M_ME_NC_1 with COA 300 IOA 12 | CLIENT OA 0
> 
> Ctrl + C to stop
> ^CStopped
> ```
> 
> Code for server and client I tested against

Thank you for providing the test code. Yes, I can confirm that the issue appears in c104==2.2.0 but works correctly in c104==2.0.1 as you mentioned.

lennart-bader, 2025-04-07:
The behavior of 2.2.0 could be related to #52 which also affects the general interrogation, i.e., only a single message is sent during the GI.

m-unkel, 2025-04-11:
We’ve addressed the problem in the latest commits, and the fix will be included in the upcoming release.
Please feel free to pull the latest changes or wait for the next release, and let us know if you encounter any further issues.

Best
Martin

m-unkel, 2025-04-15:
Release 2.2.1 is now available, and it includes the fixes that addresses this issue. I would greatly appreciate your feedback. If the issue is resolved to your satisfaction, please consider closing this thread.

Thank you!

closed 2025-04-16 by nobu121