Fraunhofer-FIT-DIEN/iec104-python #55: Support for PEP 703 / Free-threaded Python

lennart-bader, 2025-03-25:
In [PEP 703](https://peps.python.org/pep-0703) (Implemented in a beta state in Python 3.13), the GIL can be disabled to allow free-threaded Python programs.

This comes with required changes for imported C and C++ libraries to indicate that they support this GIL-free implementation.

For pybind11, this feature has been [discussed as of April 2024](https://github.com/pybind/pybind11/issues/5112) and, according to the respective [compatibility tracking issue](https://github.com/Quansight-Labs/free-threaded-compatibility/issues/13) implemented in pybind 2.13 (?).

According to the [pybind11 documentation])(https://pybind11.readthedocs.io/en/stable/reference.html), the `PYBIND11_MODULE` can now be initialized with the `py::mod_gil_not_used()` parameter to indicate that the module can be safely used without the GIL.

Are there any plans to apply this to iec104-python or maybe even already considerations that prevent an adoption of this feature?

m-unkel, 2025-04-11:
This should be a non-breaking change, so we’re planning to add experimental support and evaluate how it performs in practice.Since the change.
Please feel free to pull the latest changes or wait for the next release, and let us know if you encounter any further issues.

Best
Martin

m-unkel, 2025-04-11:
I build a package for cp313t, but the interpreter crashes on importing c104. So there is more to do then just adding this argument.

m-unkel, 2025-04-15:
I added free threaded wheels to the latest release site. Pypi seems not to accept them (502 Bad Gateway).

https://github.com/Fraunhofer-FIT-DIEN/iec104-python/releases/tag/v2.2.1