Fraunhofer-FIT-DIEN/iec104-python #56: Server and Connection timeout modify.

sintrb, 2025-03-26:
What was the purpose of the following two modifications? The default timeout for many clients is 10s. If it is adjusted to 2 seconds, it may easily cause the client to timeout.

https://github.com/Fraunhofer-FIT-DIEN/iec104-python/blob/fde09922dcb5b1b311e10cd2ac045997dd31e2df/src/Server.cpp#L103-L107
https://github.com/Fraunhofer-FIT-DIEN/iec104-python/blob/fde09922dcb5b1b311e10cd2ac045997dd31e2df/src/remote/Connection.cpp#L67-L71

m-unkel, 2025-03-26:
Thank you for bringing this to our attention. When making this change, we did not expect TCP/IP connections to be that slow, but we now recognize that reducing the timeout to 2 seconds may lead to unintended client-side timeouts.

As you are not the only one having problems with these settings, we will be reverting this change in the upcoming release to prevent any disruptions. Apologies for any issues this may have caused, and thank you for your patience and support.

Best
Martin

m-unkel, 2025-04-11:
We’ve addressed the problem in the latest commits, and the fix will be included in the upcoming release.
Please feel free to pull the latest changes or wait for the next release, and let us know if you encounter any further issues.

Best
Martin

closed 2025-05-06 by m-unkel