Fraunhofer-FIT-DIEN/iec104-python #57: Question: Should CS101_COT_REQUESTED_BY_GENERAL_COUNTER be CS101_COT_INTERROGATED_BY_STATION in interrogation response?

hezsomail, 2025-04-07:
Hello,

Thanks a lot for the excellent work on this project.

While reviewing the code in iec104-python/src/Server.cpp, I noticed something in the Server::interrogationHandler function (around line 1327), where batches are created for GI responses:

auto b = Remote::Message::Batch::create(
    CS101_COT_REQUESTED_BY_GENERAL_COUNTER);

Would it perhaps make more sense to use CS101_COT_INTERROGATED_BY_STATION here instead?

It seems like the GI replies are currently being sent with cause of transmission 37 (COT_REQUESTED_BY_GENERAL_COUNTER), but since this is the general interrogation handler, shouldn't the cause ideally be 20 (COT_INTERROGATED_BY_STATION)?

Just wanted to confirm in case this was an oversight — or if there’s some logic behind using 37 here.

Best regards,

Zsolt

lennart-bader, 2025-04-07:
Hello Zsolt,

thank you for looking into the details here. Due to renaming of the "general interrogation" into "station interrogation" in the German version of the standard, I had some trouble finding the correct part for this.

However, DIN EN 60870-5-101 covers the procedure in Section 7.4.5 and **you are absolutely right**.

| Direction | Type                       | COT                         | Description                                       |
|-----------|----------------------------|-----------------------------|---------------------------------------------------|
| C         | 100 (C_IC_NA_1)            | 6 (ACT)                     | Client initiates the general interrogation        |
| M         | 100 (C_IC_NA_1)            | 7 (ACT CON)                 | Server confirms the general interrogation command |
| M         | 1, 3, 5, 7, 11, 13, 20, 21 | 20 (INROGEN) + 21-36 (INRO X) | Server sends data                                 |
| M         | 100 (C_IC_NA_1)            | 10 (ACT TERM)               | Server ends the general interrogation             |

I'm not sure if @m-unkel has some different information or it might just be a copy & paste issue from the `Server::counterInterrogationHandler`, where COT = 37 is correct.

hezsomail, 2025-04-08:
Thank you for the quick response and for confirming the correct COT value.

I've now updated this in my own implementation accordingly.

Best regards,
Zsolt

closed 2025-04-08 by hezsomail