Fraunhofer-FIT-DIEN/iec104-python #64: TLS handhsake fails

harshvhil, 2025-08-10:
Hi all, I getting the following error when I run the code with TLS certificates : 

[image]

the code on user client and server side looks like this : 

[image]

[image]

I generated the certificates using the following script : https://github.com/Fraunhofer-FIT-DIEN/iec104-python/blob/main/bin/gen-certs.sh. 

I found another closed issue which has a similar problem marked as close, : https://github.com/Fraunhofer-FIT-DIEN/iec104-python/issues/7

But I've already tried the solutions posted there , still not working for me. Not sure what I am doing wrong.

nazim1997, 2025-11-25:
I have the exact problem

m-unkel, 2026-02-20:
Hi,
Thanks for reporting this issue.

Could you please clarify whether you’re building the project from source or using the PyPI package?

If you are building from source, this error is often caused by inconsistent or stale dependencies. Please try performing a clean dependency reset:

Remove the dependencies directory:

`rm -rf depends`

Reinitialize all submodules:

`git submodule update --init --recursive`

After that, rebuild the project and check if the issue persists.

Let us know how you installed it and which platform you’re using, and we’ll be happy to help further.

Best
Martin